tpaviot/pythonocc-core #1105: GeomLibPYTHON_wrap: Too many arguments to function call

ageeye, 2022-04-30:
If I try to build with python 3.10.2 I get this error:

`/tmp/pythonocc-core-20220430-45588-w69c6f/Build/Darwin/arm64-Release-/GeomLibPYTHON_wrap.cxx:7060:129: error: too many arguments to function call, expected at most 4, have 5
      GeomLib::FuseIntervals((NCollection_Array1< double > const &)*arg1,(NCollection_Array1< double > const &)*arg2,*arg3,arg4,arg5);
      ~~~~~~~~~~~~~~~~~~~~~~                                                                                                    ^~~~
/opt/homebrew/opt/opencascade@7.5.2/include/opencascade/GeomLib.hxx:173:31: note: 'FuseIntervals' declared here
  Standard_EXPORT static void FuseIntervals (const TColStd_Array1OfReal& Interval1, const TColStd_Array1OfReal& Interval2, TColStd_SequenceOfReal& Fusion, const Standard_Real Confusion = 1.0e-9);`

Does anyone have an idea? Thanks for support.

cafhach, 2022-05-02:
What occt version do you build it against?

ageeye, 2022-05-02:
Sorry, I thought you would seen it in the error message. It is opencascade@7.5.2 .

cafhach, 2022-05-03:
Can you build it with e.g. Python 3.9?

ageeye, 2022-05-06:
Sorry, that is not my aim.