tpaviot/pythonocc-core #1111: update to occt7.6

looooo, 2022-05-27:
@tpaviot any plans to update to occt7.6. For conda-forge occt7.6 migration there is only IfcOpenShell and pythonocc missing.

ageeye, 2022-06-19:
I started [here](https://github.com/ageeye/pythonocc-core/commits/7.6.2). Has someone information about TCollection? This header are (re)moved:

- TCollection_BaseSequence.hxx
- TCollection_BasicMap.hxx
- TCollection_BasicMapIterator.hxx
- TCollection_MapNode.hxx
- TCollection_MapNodePtr.hxx
- TCollection_SeqNode.hxx
- TCollection_SeqNodePtr.hxx
- TCollection_Side.hxx

I try to use my eyes. But I have not found in "Upgrade from older OCCT versions" a hint.

gkv311, 2022-06-20:
@ageeye, these `TCollection` header files are [remnants](https://tracker.dev.opencascade.org/view.php?id=29642) of WOK template generation, which have been dropped in OCCT 7.0.0. In the current form they are unusable for anything but legacy collections declared in CDL at application level - which is, I believe, not the case of this project.

`TColStd_PackedMapOfInteger` was the only class in OCCT itself that used legacy base class `TCollection_BasicMap`, which has been fixed in OCCT 7.6.0.

So what you need to do is just to remove inclusion of these header files.

ageeye, 2022-06-21:
@gkv311 Thank you for your fast answer. That sounds good. After work I will test it.

ageeye, 2022-06-21:
It works. But if one issue is solved, two new problems are born.

Krande, 2022-08-07:
FWIW a while back I also experimented with moving pythonocc-core to v7.6 by modifying the [pythonocc-generator](https://github.com/Krande/pythonocc-generator) that updates the src directory in https://github.com/Krande/pythonocc-core/tree/v7.6.2a1 

I also set up a github actions pipeline to automatically compile pythonocc-core on each push for win, linux and osx. I haven't been successfully in my attempts yet. But maybe if we combine our efforts we might be able to update pythonocc-core to support v7.6?

tpaviot, 2022-08-17:
@looooo  wrapper has been updated to occt-7.6.2, the pythonocc-demos updated. Release 7.6.2 tagged and ready to be deployed at conda-forge

looooo, 2022-08-17:
thanks, https://github.com/conda-forge/pythonocc-core-feedstock/pull/14

tpaviot, 2022-08-23:
@looooo that's great, thank you, this issue can be closed

closed 2022-08-23 by tpaviot