tpaviot/pythonocc-core #1135: Exporting STEP to THREEJS --> Assembly structure lost in JSON file

freddida, 2022-08-23:
Hello pythonocc-community,

**Issue**
I want to export a step file as a json string to use it in threejs using the following code:

    step_reader = STEPControl_Reader()
    status = step_reader.ReadFile(inputFile)

    if status == IFSelect_RetDone:  # check status
        failsonly = False
        step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
        step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)

        step_reader.TransferRoot(1)
        _nbs = step_reader.NbShapes()
        _shape = step_reader.Shape(1)
        _tess = ShapeTesselator(_shape)
        _tess.Compute(compute_edges=False, mesh_quality=1, parallel=True)

        with open(outputFile, "w") as text_file:
            json = _tess.ExportShapeToThreejsJSONString(inputFile)
            json = json.replace("\\", "/")
            text_file.write(json)
    else:
        raise Exception("Error: can't read file - Method: _load_STEP_file"))`

The json file is working fine but the file structure of the step is lost.
File: [turbine_stp.zip](https://github.com/tpaviot/pythonocc-core/files/9401826/turbine_stp.zip)

**Expected behavior**
The json string should contain the threejs structure (e.g. different parts, ect.)

**Question**
Is there a possibility to achieve this?

 - Version: [latest]

gkv311, 2022-08-30:
Just curious, why not converting STEP document into glTF file using `STEPCAFControl_Reader` and `RWGltf_CafWriter`? It would try to keep assembly structure, and Three.js has glTF reader.

[image]

freddida, 2022-08-30:
Hey,

Thank you! I was not aware of these functions!
Would you be so kind to provide the code you used for this output?