tpaviot/pythonocc-core #1154: Is there a way to change an exports HEADER section?

Jeremy-Walton, 2022-11-11:
I am exporting a STEP file using this library, but am trying to figure out how to set the header info like File Description, Author, File name, etc.

My current file has
```
HEADER;
FILE_DESCRIPTION(('Open CASCADE Model'),'2;1');
FILE_NAME('Open CASCADE Shape Model','2022-11-10T20:06:38',('Author'),(
    'Open CASCADE'),'Open CASCADE STEP processor 7.6','Open CASCADE 7.6'
  ,'Unknown');
FILE_SCHEMA(('AUTOMOTIVE_DESIGN { 1 0 10303 214 1 1 1 1 }'));
ENDSEC;
```

I can use `STEPControl_Writer().Model().ClearHeader()` to remove everything in that section, but I cannot figure out how to change it.

In https://dev.opencascade.org/doc/occt-7.6.0/refman/html/class_step_data___step_model.html#aad41ab9f8467bf843649fdd63040c021 there is a `AddHeaderEntity` method on the Model which can be called with `STEPControl_Writer().Model().AddHeaderEntity()` but I cannot figure out how to construct the entity it expects or use it.

I also see https://dev.opencascade.org/doc/occt-7.6.0/refman/html/class_a_p_i_header_section___make_header.html which seems to do exactly what I want, but is not exposed in the python library.

Here is a simplified version of what I'm doing:

```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.STEPControl import STEPControl_Writer, STEPControl_AsIs
from OCC.Core.Interface import Interface_Static_SetCVal
from OCC.Core.IFSelect import IFSelect_RetDone

# creates a basic shape
box_s = BRepPrimAPI_MakeBox(10, 20, 30).Shape()

# initialize the STEP exporter
step_writer = STEPControl_Writer()
dd = step_writer.WS().TransferWriter().FinderProcess()
print(dd)

Interface_Static_SetCVal("write.step.schema", "AP203")

# transfer shapes and write file
Interface_Static_SetCVal('write.step.product.name', 'Box')
step_writer.Transfer(box_s, STEPControl_AsIs)

# step_writer.Model().ClearHeader()
# step_writer.Model().AddHeaderEntity(whatdoiputhere?)

status = step_writer.Write("box.stp")

if status != IFSelect_RetDone:
    raise AssertionError("load failed")
```

Jeremy-Walton, 2022-11-11:
Here is the full file

```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION(('Open CASCADE Model'),'2;1');
FILE_NAME('Open CASCADE Shape Model','2022-11-10T20:20:26',('Author'),(
    'Open CASCADE'),'Open CASCADE STEP processor 7.6','Open CASCADE 7.6'
  ,'Unknown');
FILE_SCHEMA(('AUTOMOTIVE_DESIGN { 1 0 10303 214 1 1 1 1 }'));
ENDSEC;
DATA;
#1 = APPLICATION_PROTOCOL_DEFINITION('international standard',
  'config_control_design',1994,#2);
#2 = APPLICATION_CONTEXT(
'configuration controlled 3D designs of mechanical parts and assemblies'
  );
#3 = SHAPE_DEFINITION_REPRESENTATION(#4,#10);
#4 = PRODUCT_DEFINITION_SHAPE('','',#5);
#5 = PRODUCT_DEFINITION('design','',#6,#9);
#6 = PRODUCT_DEFINITION_FORMATION_WITH_SPECIFIED_SOURCE('','',#7,
  .NOT_KNOWN.);
#7 = PRODUCT('Box 1','Box 1','',(#8));
#8 = MECHANICAL_CONTEXT('',#2,'mechanical');
#9 = DESIGN_CONTEXT('',#2,'design');
#10 = ADVANCED_BREP_SHAPE_REPRESENTATION('',(#11,#15),#345);
#11 = AXIS2_PLACEMENT_3D('',#12,#13,#14);
#12 = CARTESIAN_POINT('',(0.,0.,0.));
#13 = DIRECTION('',(0.,0.,1.));
#14 = DIRECTION('',(1.,0.,-0.));
#15 = MANIFOLD_SOLID_BREP('',#16);
#16 = CLOSED_SHELL('',(#17,#137,#237,#284,#331,#338));
#17 = ADVANCED_FACE('',(#18),#32,.F.);
#18 = FACE_BOUND('',#19,.F.);
#19 = EDGE_LOOP('',(#20,#55,#83,#111));
#20 = ORIENTED_EDGE('',*,*,#21,.F.);
#21 = EDGE_CURVE('',#22,#24,#26,.T.);
#22 = VERTEX_POINT('',#23);
#23 = CARTESIAN_POINT('',(0.,0.,0.));
#24 = VERTEX_POINT('',#25);
#25 = CARTESIAN_POINT('',(0.,0.,30.));
#26 = SURFACE_CURVE('',#27,(#31,#43),.PCURVE_S1.);
#27 = LINE('',#28,#29);
#28 = CARTESIAN_POINT('',(0.,0.,0.));
#29 = VECTOR('',#30,1.);
#30 = DIRECTION('',(0.,0.,1.));
#31 = PCURVE('',#32,#37);
#32 = PLANE('',#33);
#33 = AXIS2_PLACEMENT_3D('',#34,#35,#36);
#34 = CARTESIAN_POINT('',(0.,0.,0.));
#35 = DIRECTION('',(1.,0.,-0.));
#36 = DIRECTION('',(0.,0.,1.));
#37 = DEFINITIONAL_REPRESENTATION('',(#38),#42);
#38 = LINE('',#39,#40);
#39 = CARTESIAN_POINT('',(0.,0.));
#40 = VECTOR('',#41,1.);
#41 = DIRECTION('',(1.,0.));
#42 = ( GEOMETRIC_REPRESENTATION_CONTEXT(2) 
PARAMETRIC_REPRESENTATION_CONTEXT() REPRESENTATION_CONTEXT('2D SPACE',''
  ) );
#43 = PCURVE('',#44,#49);
#44 = PLANE('',#45);
#45 = AXIS2_PLACEMENT_3D('',#46,#47,#48);
#46 = CARTESIAN_POINT('',(0.,0.,0.));
#47 = DIRECTION('',(-0.,1.,0.));
#48 = DIRECTION('',(0.,0.,1.));
#49 = DEFINITIONAL_REPRESENTATION('',(#50),#54);
#50 = LINE('',#51,#52);
#51 = CARTESIAN_POINT('',(0.,0.));
#52 = VECTOR('',#53,1.);
[... 356 lines cut ...]
#359 = PERSON_AND_ORGANIZATION_ROLE('design_owner');
#360 = CC_DESIGN_PERSON_AND_ORGANIZATION_ASSIGNMENT(#354,#361,(#6));
#361 = PERSON_AND_ORGANIZATION_ROLE('design_supplier');
#362 = CC_DESIGN_PERSON_AND_ORGANIZATION_ASSIGNMENT(#354,#363,(#364));
#363 = PERSON_AND_ORGANIZATION_ROLE('classification_officer');
#364 = SECURITY_CLASSIFICATION('','',#365);
#365 = SECURITY_CLASSIFICATION_LEVEL('unclassified');
#366 = CC_DESIGN_SECURITY_CLASSIFICATION(#364,(#6));
#367 = CC_DESIGN_DATE_AND_TIME_ASSIGNMENT(#368,#372,(#5));
#368 = DATE_AND_TIME(#369,#370);
#369 = CALENDAR_DATE(2022,10,11);
#370 = LOCAL_TIME(20,20,$,#371);
#371 = COORDINATED_UNIVERSAL_TIME_OFFSET(5,$,.BEHIND.);
#372 = DATE_TIME_ROLE('creation_date');
#373 = CC_DESIGN_DATE_AND_TIME_ASSIGNMENT(#368,#374,(#364));
#374 = DATE_TIME_ROLE('classification_date');
#375 = CC_DESIGN_APPROVAL(#376,(#6,#5,#364));
#376 = APPROVAL(#377,'');
#377 = APPROVAL_STATUS('not_yet_approved');
#378 = APPROVAL_PERSON_ORGANIZATION(#354,#376,#379);
#379 = APPROVAL_ROLE('approver');
#380 = APPROVAL_DATE_TIME(#368,#376);
ENDSEC;
END-ISO-10303-21;
```

LuisGMM, 2023-01-09:
Hi @Jeremy-Walton , I am in the exact same situation. Did you find any solution or workaround to this?

Jeremy-Walton, 2023-01-09:
@LuisGMM i ended up switching to use https://github.com/donalffons/opencascade.js-examples since it integrated with our stack better, but we didn’t solve the problem their either so it’s still an open question.

tpaviot, 2023-01-10:
I added a few missing classes to the python wrapper to enable step header setting/modification (see commit 44e8924). The modified code to change the header is:
```python
from datetime import datetime

from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.STEPControl import STEPControl_Writer, STEPControl_AsIs
from OCC.Core.Interface import Interface_Static_SetCVal
from OCC.Core.IFSelect import IFSelect_RetDone

from OCC.Core.Interface import Interface_HArray1OfHAsciiString
from OCC.Core.APIHeaderSection import APIHeaderSection_MakeHeader
from OCC.Core.TCollection import TCollection_HAsciiString

# creates a basic shape
box_s = BRepPrimAPI_MakeBox(10, 20, 30).Shape()

# initialize the STEP exporter
step_writer = STEPControl_Writer()
dd = step_writer.WS().TransferWriter().FinderProcess()

Interface_Static_SetCVal("write.step.schema", "AP203")

# transfer shapes and write file
Interface_Static_SetCVal("write.step.product.name", "Box")
step_writer.Transfer(box_s, STEPControl_AsIs)

#
# Set STEP header
#
model = step_writer.Model()
model.ClearHeader()

hs = APIHeaderSection_MakeHeader()
hs.SetName(TCollection_HAsciiString("model name"))
hs.SetAuthorValue(1, TCollection_HAsciiString("My name"))
hs.SetAuthorisation(TCollection_HAsciiString("authorization"))

descr = Interface_HArray1OfHAsciiString(1, 3)
descr.SetValue(1, TCollection_HAsciiString("a description"))
descr.SetValue(2, TCollection_HAsciiString("splitted into"))
descr.SetValue(3, TCollection_HAsciiString("three items"))
hs.SetDescription(descr)

org = Interface_HArray1OfHAsciiString(1, 1)
org.SetValue(1, TCollection_HAsciiString("pythonocc organization"))
hs.SetOrganization(org)

hs.SetOriginatingSystem(TCollection_HAsciiString("pythonocc originating system"))
hs.SetImplementationLevel(TCollection_HAsciiString("implementation level"))

identifiers = Interface_HArray1OfHAsciiString(1, 1)
identifiers.SetValue(1, TCollection_HAsciiString("a schema identifier"))
hs.SetSchemaIdentifiers(identifiers)

hs.SetPreprocessorVersion(TCollection_HAsciiString("preprocessor version"))
hs.SetTimeStamp(TCollection_HAsciiString(f"Time stamp: {datetime.now()}"))

model.AddHeaderEntity(hs.FnValue())
model.AddHeaderEntity(hs.FsValue())
model.AddHeaderEntity(hs.FdValue())

# finally write file
status = step_writer.Write("box.stp")

if status != IFSelect_RetDone:
    raise AssertionError("load failed")
```

and the related box.stp output becomes
```bash
ISO-10303-21;
HEADER;
FILE_NAME('model name','Time stamp: 2023-01-10 04:09:47.999706',(
    'My name'),('pythonocc organization'),'preprocessor version',
  'pythonocc originating system','authorization');
FILE_SCHEMA(('a schema identifier'));
FILE_DESCRIPTION(('a description','splitted into','three items'),
  'implementation level');
ENDSEC;
DATA;
#1 = APPLICATION_PROTOCOL_DEFINITION('international standard',
  'config_control_design',1994,#2);
#2 = APPLICATION_CONTEXT(
'configuration controlled 3D designs of mechanical parts and assemblies'
  );
[...]
```

tpaviot, 2023-01-10:
Example added to the pythonocc-demos suite https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_export_step_header_section.py

LuisGMM, 2023-01-10:
@Jeremy-Walton I didn't know of that library. Thanks for answering and sharing it :). 

@tpaviot Thank you very much for the speed of the answer and the snippet!