tpaviot/pythonocc-core #1155: SimpleGui visualization failed

whjdark, 2022-11-14:
I used SimpleGui to visualize a shape.
`display, start_display, add_menu, add_function_to_menu = init_display()

    ais_shp = AIS_ColoredShape(shape)
    for face in bottom_map.keys():
        if bottom_map[face]:
            ais_shp.SetCustomColor(face, rgb_color(0,1,0))

    display.Context.Display(ais_shp, True)
    display.FitAll()

    start_display()`
Howerver, i got a wrong visualization, which lost a lot of faces.
[image]
I tried to save the step file and used STP viewer to visualize. The visualization was correct.
[image]

Tanneguydv, 2023-03-13:
Have you tried this?

```
from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

display.DisplayShape(shape)
display.FitAll()
start_display()
```

di-huang, 2023-05-27:
> Have you tried this?
> 
> ```
> from OCC.Display.SimpleGui import init_display
> display, start_display, add_menu, add_function_to_menu = init_display()
> 
> display.DisplayShape(shape)
> display.FitAll()
> start_display()
> ```

Thank you for your reply. I'm wondering how to display a 3D model with white surface and black edge lines using SimpleGui, and also how to adjust line width. Looking forward to your reply. Thank you very much.