tpaviot/pythonocc-core #1158: I have get trouble in the use "XCAFDoc_Dimension",I think their are something wrong .

ealwen, 2022-11-25:
My pythonocc's version is 7.6.2
when I try to get the PMI message(GB&T) stored in my step file,I get the wrong like this:
>TypeError: Wrong number or type of arguments for overloaded function 'TDF_Label_FindAttribute'.
  Possible C/C++ prototypes are:
    TDF_Label::FindAttribute(Standard_GUID const &,opencascade::handle< TDF_Attribute > &)
    TDF_Label::FindAttribute(Standard_GUID const &,Standard_Integer const,opencascade::handle< TDF_Attribute > &)

I find their is someting wrong in the class <kbd>XCAFDoc_Dimension</kbd>  ，it inherite the class <kbd>object</kbd> wrongly,I think it should  nherite the class <kbd>OCC.Core.TDF.TDF_Attribute</kbd> and there is something wrong about class <kbd>XCAFDoc_GeomTolerance</kbd>,it show :<kbd>@classnotwrapped</kbd>
the test-code like this ( my code) :

```python

#!/usr/bin/env python3

from OCC.Core.STEPCAFControl import STEPCAFControl_Reader
from OCC.Core.TCollection import TCollection_ExtendedString
from OCC.Core.TDF import TDF_LabelSequence
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.IFSelect import IFSelect_RetDone
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool_DimTolTool, XCAFDoc_Dimension,XCAFDoc_Datum,XCAFDoc_Dimension_GetID,XCAFDoc_DocumentTool,XCAFDoc_DimTol,XCAFDoc_GeomTolerance
doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))

filename=r"...\nist_ctc_05.stp"  #my step file 
step_reader = STEPCAFControl_Reader()
step_reader.SetColorMode(True)
step_reader.SetLayerMode(True)
step_reader.SetNameMode(True)
step_reader.SetMatMode(True)
step_reader.SetGDTMode(True)

status = step_reader.ReadFile(filename)

doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))
step_reader.Transfer(doc)

if status == IFSelect_RetDone:
    step_reader.Transfer(doc)

#PMI,GB&T工具
m_dimTolTool=XCAFDoc_DocumentTool.DimTolTool(doc.Main())
m_dimTolTool=XCAFDoc_DocumentTool_DimTolTool(doc.Main())

#尺寸
DimsionLabels=TDF_LabelSequence()

m_dimTolTool.GetDimensionLabels(DimsionLabels)
print(DimsionLabels.Length())

for i in range(DimsionLabels.Length()):
    aLabel=DimsionLabels.Value(i+1)
    print(aLabel.DumpToString())

    aDimAttr=XCAFDoc_Dimension()
    # aDimAttr.Set(aLabel)
    # aDimAttr.GetObject()
    # aDimTolObj = aDimAttr.GetObject()
    if not aLabel.FindAttribute(XCAFDoc_Dimension.GetID(),aDimAttr):  #the wrong line
        continue
    aDimTolObj=aDimAttr.GetObject()

```

aruig666, 2024-05-05:
I have encountered the same problem, may I ask if you have solved it？

aruig666, 2024-05-24:
I have the same problem. Have you solved it?