tpaviot/pythonocc-core #1161: Building on M1 MBPro

rckirby, 2022-12-06:
I thought I had successfully built pythonocc-core on my MBPro with dependencies installed via homebrew.
(I'm using this inside of Firedrake, which doesn't work well with conda)

I can import the top-level python libraries, but an import fails deep inside trying to use it.  I've bumped up the import to the top level Python interpreter.  How to trouble-shoot?

```
IPython 8.7.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from OCC.Core.TopExp import topexp_MapShapesAndAncestors
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[1], line 1
----> 1 from OCC.Core.TopExp import topexp_MapShapesAndAncestors

ImportError: cannot import name 'topexp_MapShapesAndAncestors' from 'OCC.Core.TopExp' (/Users/robert_kirby/Code/firedrake/lib/python3.10/site-packages/OCC/Core/TopExp.py)
```

tpaviot, 2022-12-18:
@rckirby try the following:

```python
from OCC.Core.TopExp import topexp
```
and use:
```python
topexp.MapShapesAndAncestors(sthg)
```

vincentmarchetti, 2023-03-25:
I see this type of error when running the test code after what appeared to be a successful build from source of OpenCascade 7.4.0 and pyocc-core 7.4.1 on a Mac with M1 CPU, running OS 13.2.1 (Ventura), building with tools in Xcode 14.2

```
(groucho) pythonocc-core-7.4.1/test: python run_tests.py
Traceback (most recent call last):
  File "/Users/vmarchetti/Documents/attic/oc_sw_minnie/build_occt/build_pyocc/pythonocc-core-7.4.1/test/run_tests.py", line 22, in <module>
    import core_wrapper_features_unittest
  File "/Users/vmarchetti/Documents/attic/oc_sw_minnie/build_occt/build_pyocc/pythonocc-core-7.4.1/test/core_wrapper_features_unittest.py", line 37, in <module>
    from OCC.Core.BRepBndLib import brepbndlib_Add
ImportError: cannot import name 'brepbndlib_Add' from 'OCC.Core.BRepBndLib' (/Users/vmarchetti/groucho/occt/lib/python3/site-packages/OCC/Core/BRepBndLib.py)
(groucho) pythonocc-core-7.4.1/test:  
```
Is code in the pythonocc-core-7.4.1/test folder (and corresponding folders in later pythonocc-core releases) still relevant?

vincentmarchetti, 2023-03-25:
So to answer my own question -- the tests in the test folder for the pythonocc-core source distributions are still relevant, in that I can run them in the conda installation of pyocc and OpenCascade, and they run and succeed. Problem is apparently in the build from source.

vincentmarchetti, 2023-03-27:
By comparing the successful installation in conda and my unsuccessful buil of pyocc-core-7-7-0 I have discovered that the problem may be the use of SWIG 4.1.0 in my own build. The latest conda installation has code generated by SWIG 4.0.2 and the generated  code for BRepBndLib.py built by 4.0.2 contains an explicit definition for a function brepbndlib_Add while the BRepBndLib.py generated by SWIG 4.1.1 does not. I plan to redo my build with SWIG 4.0.2 in the coming days and will report on result.

tpaviot, 2023-03-27:
@vincentmarchetti yes, swig-4.1 is not currently supported, you have to use swig-4.0.2.

closed 2023-04-19 by tpaviot