tpaviot/pythonocc-core #1167: DisplayMessage

thnszgns, 2022-12-21:
I have some issues while trying to show text in my viewer.. End goal is to have a solid and show one number in each vertex (1,2,3,......)
I currently do it like this:
_self.display.DisplayMessage(gp_Pnt(10, 10, 10), 1)_

My problems is that I have tried everything I found but there seems to be no way to delete all these messages.. I am trying _self.display.EraseAll()_  and everything that has to do with erase and Remove in the display, viewer or context but nothing works. All the other shapes showed with DisplayShape are successfully removed

tnakaicode, 2022-12-23:
I have same problem.
I solve this way:

```
text = display.DisplayMessage(gp_Pnt(0,0,0), "text0")
...
text.Clear()
text = display.DisplayMessage(gp_Pnt(0,0,0), "text1")
...
```

```display.DisplayMessage()``` return ```Graphic3d_Structure()```.

thnszgns, 2023-01-10:
ok solved! Thank you very much for your time!

closed 2023-01-10 by thnszgns