tpaviot/pythonocc-core #1176: free(): invalid pointer

sfornengo, 2023-02-12:
Here is the code
`from OCC.Core.gp import gp_Pnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSpline
from OCC.Core.TColgp import TColgp_Array1OfPnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.Core.BRepAdaptor import BRepAdaptor_Curve

# the bspline
array = TColgp_Array1OfPnt(1, 5)
array.SetValue(1, gp_Pnt(0,0,1))
array.SetValue(2, gp_Pnt(0,0,2))
array.SetValue(3, gp_Pnt(0,0,3))
array.SetValue(4, gp_Pnt(0,0,4))
array.SetValue(5, gp_Pnt(0,0,5))
bspline = GeomAPI_PointsToBSpline(array).Curve()
instance = BRepBuilderAPI_MakeEdge(bspline).Edge()

adap=BRepAdaptor_Curve(instance)
crv = adap.Curve()`

free(): invalid pointer
Aborted (core dumped)

tpaviot, 2023-02-13:
@sfornengo points lie on a straight line, you can't use a BSpline to interpolate those points

sfornengo, 2023-02-13:
Ok thanks, this points was manually taken for isolating the bug in a reduced code, sorry for the alignment.
But points not lying on a straight line lead to the same error too.

tpaviot, 2024-05-23:
Fixed by PR #1334