tpaviot/pythonocc-core #1177: RuntimeError: out of range index raised from method Compute of class ShapeTesselator

paihui, 2023-02-13:
I test the code from core_tesselator_unittest.py and I try to test my file, I keep getting the following runtime error: Standard_OutOfRangeNCollection_AliasedArray::value(), out of range index raised from method Compute of class ShapeTesselator.
It does not seem to be coming from my code. I have no idea how to fix it or check it.

You can find my code below and the full terminal output

- my test code

[image]

- terminal  output

ERROR: test_tesselate_STEP_file (__main__.TestTesselator)
loads a step file, tesselate. The as1_pe_203 contains
----------------------------------------------------------------------
Traceback (most recent call last):
  File "d:\pythonocc-core\test\core_tesselator_unittest.py", line 129, in test_tesselate_STEP_file
    stp_file_tesselator.Compute(compute_edges=True)
  File "C:\Anaconda3\envs\py3d\lib\site-packages\OCC\Core\Tesselator.py", line 250, in Compute
    return _Tesselator.ShapeTesselator_Compute(self, compute_edges, mesh_quality, parallel)
RuntimeError: Standard_OutOfRangeNCollection_AliasedArray::value(), out of range index raised from method Compute of class ShapeTesselator

----------------------------------------------------------------------
Ran 8 tests in 1.324s

FAILED (errors=1)

tpaviot, 2023-02-13:
Can you please provide the stp file you used for your test

paihui, 2023-02-14:
I used two files for my test, one is as1-oc-214.stp in pythonocc-core/test/test_io/.
[test.zip](https://github.com/tpaviot/pythonocc-core/files/10728731/test.zip)

ChristophNa, 2023-03-23:
The same problem occurs for pythonocc-demos/examples/core_webgl_STEP_to_X3D.py and pythonocc-demos/examples/core_webgl_x3dom_bigfile_multipleshapes.py

RuntimeError: Standard_OutOfRangeNCollection_AliasedArray::value(), out of range index raised from method Compute of class ShapeTesselator

I used the step files provided in the repository. 
I used the following version: pythonocc-core 7.7.0

wrxsky, 2023-04-12:
I also get the same error when running "load_step_ap203_multiple_shapes.ipynb" in /jupyter_notebooks.  With pythonocc-core 7.7.0, Jupyter Notebook 6.5.4 and Python 3.10.10.