tpaviot/pythonocc-core #1180: `OffscreenRenderer` doesn't propagate `size` correctly.

gestj, 2023-03-02:
The function `init_display ` provides a `size` parameter which is meant for the screen size of the created and returned display/renderer (however you wanna call it).

BUT, in case of offscreen rendering (`env. PYTHONOCC_OFFSCREEN_RENDERER = 1`) the `size` isn't propagated correctly. Instead a default of 640x480 will be used ALWAYS. This is unexpected.

I use now my own implementation which is more or less c&p of your `OffscreenRenderer` including overwriting the `Viewer3d` and there the `InitOffscreen` method because that was the most straight forward hack for me.

In `Viewer3d.Create` https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/OCCViewer.py#L203 you can see that even if `OffscreenRenderer` would consider `size` correctly (in fact it does consider it a "little" bit; see its constructor) it wouldn't work as expected.

Overall the code is just wired incorrectly for this use case...

I might be able to provide a fix if you want, but, I am unsure how activly this entire repo is maintained anyway... So I won't spend time upfront on this since I am very very very busy with my own stuff.

tpaviot, 2023-03-03:
feel free to open a PR. Since you already patched related file(s), this should not take too much of your precious precious precious time (15mn?). Anyway, this is two or three orders of magnitude lower than the overall time I spent providing this library for free. Honestly, no one here cares about your agenda.

Tanneguydv, 2023-03-03:
In the readme section we can read : "Latest release: [pythonocc-core 7.7.0 (December 2022)](https://github.com/tpaviot/pythonocc-core/releases/tag/7.7.0)"
one could guess that this "entire repo" is maintained ...