tpaviot/pythonocc-core #1181: (Bug?) XCAFDoc_Material_Tool not working when reading materials

LuisGMM, 2023-03-07:
XCAFDoc_Material_Tool seems to have unexpected behaviors.

I have modified [OCC.Extend.DataExchange.read_with_attributes_step](https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py#L143) to read materials using XCAFDoc_Material_Tool in the following manner (this is only the important bit):

```python

def read_with_attributes_step(filename: str):

    # Create an handle to a document
    doc = TDocStd_Document(TCollection_ExtendedString('File Document'))

    # Get root assembly
    shape_tool: XCAFDoc_ShapeTool = XCAFDoc_DocumentTool_ShapeTool(doc.Main())
    color_tool: XCAFDoc_ColorTool = XCAFDoc_DocumentTool_ColorTool(doc.Main())
    mat_tool: XCAFDoc_MaterialTool = XCAFDoc_DocumentTool_MaterialTool(doc.Main())

    # By default all modes are set to True (SetColorMode, SetGDTMode,SetMatMode, SetPropsMode ...)
    step_reader = STEPCAFControl_Reader()
    status = step_reader.ReadFile(filename)

    if status == IFSelect_RetDone:
        step_reader.Transfer(doc)
    else:
        log_step_reader.error(f'Unexpected read status: {status}')

    labels = TDF_LabelSequence()
    shape_tool.GetFreeShapes(labels)

    material_labels = TDF_LabelSequence()
    mat_tool.GetMaterialLabels(material_labels)

    for i in range(1, material_labels.Length()+1):

        material_name = TCollection_HAsciiString('hello')
        material_description = TCollection_HAsciiString('hello')
        material_densname = TCollection_HAsciiString('hello')
        material_densvaltype = TCollection_HAsciiString('hello')

        print(
            mat_tool.GetMaterial(
                material_labels.Value(i),
                material_name,
                material_description,
                material_densname,
                material_densvaltype
            ),
            material_name.ToCString(),
            material_densname.ToCString(),
            material_description.ToCString(),
            material_densvaltype.ToCString()
        )
```

The output with [this file](https://drive.google.com/file/d/1SjfM8LHB5vc8olHICfCWlfhK5W_T1aDV/view?usp=sharing) (Eight cylinders symmetrically rotated, each with a material) is:
```bash
(True, 7.85e-06) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
(True, 0.0) hello hello hello hello
```

**which means the function is not only not working as expected but not working at all**, because that is not the return type of the OCCT's [XCAFDoc_Material_Tool.Get_Material function](https://dev.opencascade.org/doc/refman/html/class_x_c_a_f_doc___material_tool.html). I add here the cpp code of that function (that evidently doesn't return a tuple):
```cpp
Standard_Boolean XCAFDoc_MaterialTool::GetMaterial(const TDF_Label& MatL,
                                                   Handle(TCollection_HAsciiString)& aName,
                                                   Handle(TCollection_HAsciiString)& aDescription,
                                                   Standard_Real& aDensity,
                                                   Handle(TCollection_HAsciiString)& aDensName,
                                                   Handle(TCollection_HAsciiString)& aDensValType) const
{
  Handle(XCAFDoc_Material) MatAttr;
  if(!MatL.FindAttribute(XCAFDoc_Material::GetID(),MatAttr)) {
    return Standard_False;
  }
  aName = MatAttr->GetName();
  aDescription = MatAttr->GetDescription();
  aDensity = MatAttr->GetDensity();
  aDensName = MatAttr->GetDensName();
  aDensValType = MatAttr->GetDensValType();
  
  return Standard_True;
}
```
I have also tried to implement that functionality with [XCAF_Doc_Material](https://dev.opencascade.org/doc/refman/html/class_x_c_a_f_doc___material.html), without succes. This is the modified read function that I used:
```python

def read_with_attributes_step(filename: str):

    # Create an handle to a document
    doc = TDocStd_Document(TCollection_ExtendedString('File Document'))

    # Get root assembly
    shape_tool: XCAFDoc_ShapeTool = XCAFDoc_DocumentTool_ShapeTool(doc.Main())
    color_tool: XCAFDoc_ColorTool = XCAFDoc_DocumentTool_ColorTool(doc.Main())
    mat_tool: XCAFDoc_MaterialTool = XCAFDoc_DocumentTool_MaterialTool(doc.Main())

    # By default all modes are set to True (SetColorMode, SetGDTMode,SetMatMode, SetPropsMode ...)
    step_reader = STEPCAFControl_Reader()
    status = step_reader.ReadFile(filename)

    if status == IFSelect_RetDone:
        step_reader.Transfer(doc)
    else:
        log_step_reader.error(f'Unexpected read status: {status}')

    labels = TDF_LabelSequence()
    shape_tool.GetFreeShapes(labels)

    material_labels = TDF_LabelSequence()
    mat_tool.GetMaterialLabels(material_labels)

    for i in range(1, material_labels.Length()+1):

        mat_attr = XCAFDoc_Material()
        mat_label = material_labels.Value(i)
        mat_label.FindAttribute(XCAFDoc_Material.GetID(), mat_attr)
        
        print('mat_attr:')
        print(mat_attr.GetDensity(),
              mat_attr.GetName(),
              mat_attr.GetDensName(),
              mat_attr.GetDensValType(),
              mat_attr.GetDescription())
```

And this is the deceiving output:
```bash
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
mat_attr:
0.0 None None None None
```
Note that the `.ToCString()` method should be called by the getter methods, but they return None...

I am not sure where the real issue is, whether it is a OCCT bug or something strange is happening in pythonocc-core.
Any idea on how to solve it/proceed?

If solved, maybe this a good oportunity to add import/export materials support in `DataExchange` (I could do a few commits with some code I have :)

LuisGMM, 2023-03-07:
I add the step contents of the file here:

```
ISO-10303-21;
HEADER;
FILE_NAME('8cyl_with_headers','Time stamp: 2023-03-07 07:33:29.671877',(
    'Radian Systems'' automated servers'),('Open CASCADE'),
  'Linux machine','Radian Systems'' servers','spain@radian.systems');
FILE_SCHEMA(('AP214IS'));
FILE_DESCRIPTION(('Radian Systems'),'3');
ENDSEC;
DATA;
#1 = APPLICATION_PROTOCOL_DEFINITION('international standard',
  'automotive_design',2000,#2);
#2 = APPLICATION_CONTEXT(
  'core data for automotive mechanical design processes');
#3 = SHAPE_DEFINITION_REPRESENTATION(#4,#10);
#4 = PRODUCT_DEFINITION_SHAPE('','',#5);
#5 = PRODUCT_DEFINITION('design','',#6,#9);
#6 = PRODUCT_DEFINITION_FORMATION('','',#7);
#7 = PRODUCT('8Cylinders v2','8Cylinders v2','',(#8));
#8 = PRODUCT_CONTEXT('',#2,'mechanical');
#9 = PRODUCT_DEFINITION_CONTEXT('part definition',#2,'design');
#10 = SHAPE_REPRESENTATION('',(#11,#15),#19);
#11 = AXIS2_PLACEMENT_3D('',#12,#13,#14);
#12 = CARTESIAN_POINT('',(0.,0.,0.));
#13 = DIRECTION('',(0.,0.,1.));
#14 = DIRECTION('',(1.,0.,-0.));
#15 = AXIS2_PLACEMENT_3D('',#16,#17,#18);
#16 = CARTESIAN_POINT('',(0.,0.,0.));
#17 = DIRECTION('',(0.,0.,1.));
#18 = DIRECTION('',(1.,0.,-0.));
#19 = ( GEOMETRIC_REPRESENTATION_CONTEXT(3) 
GLOBAL_UNCERTAINTY_ASSIGNED_CONTEXT((#23)) GLOBAL_UNIT_ASSIGNED_CONTEXT(
(#20,#21,#22)) REPRESENTATION_CONTEXT('Context #1',
  '3D Context with UNIT and UNCERTAINTY') );
#20 = ( LENGTH_UNIT() NAMED_UNIT(*) SI_UNIT(.MILLI.,.METRE.) );
#21 = ( NAMED_UNIT(*) PLANE_ANGLE_UNIT() SI_UNIT($,.RADIAN.) );
#22 = ( NAMED_UNIT(*) SI_UNIT($,.STERADIAN.) SOLID_ANGLE_UNIT() );
#23 = UNCERTAINTY_MEASURE_WITH_UNIT(LENGTH_MEASURE(1.E-07),#20,
  'distance_accuracy_value','confusion accuracy');
#24 = PRODUCT_RELATED_PRODUCT_CATEGORY('part',$,(#7));
#25 = SHAPE_DEFINITION_REPRESENTATION(#26,#32);
#26 = PRODUCT_DEFINITION_SHAPE('','',#27);
#27 = PRODUCT_DEFINITION('design','',#28,#31);
#28 = PRODUCT_DEFINITION_FORMATION('','',#29);
#29 = PRODUCT('SOLID','SOLID','',(#30));
#30 = PRODUCT_CONTEXT('',#2,'mechanical');
#31 = PRODUCT_DEFINITION_CONTEXT('part definition',#2,'design');
#32 = ADVANCED_BREP_SHAPE_REPRESENTATION('',(#11,#33),#131);
#33 = MANIFOLD_SOLID_BREP('',#34);
#34 = CLOSED_SHELL('',(#35,#123,#127));
#35 = ADVANCED_FACE('',(#36),#49,.T.);
#36 = FACE_BOUND('',#37,.T.);
#37 = EDGE_LOOP('',(#38,#72,#95,#122));
#38 = ORIENTED_EDGE('',*,*,#39,.F.);
#39 = EDGE_CURVE('',#40,#40,#42,.T.);
#40 = VERTEX_POINT('',#41);
#41 = CARTESIAN_POINT('',(211.65287853202,95.23152229471,206.46570424998
    ));
#42 = SURFACE_CURVE('',#43,(#48,#60),.PCURVE_S1.);
#43 = CIRCLE('',#44,25.);
#44 = AXIS2_PLACEMENT_3D('',#45,#46,#47);
#45 = CARTESIAN_POINT('',(229.33054806168,77.553852765046,
    206.46570424998));
#46 = DIRECTION('',(0.57922796534,0.57922796534,0.573576436351));
#47 = DIRECTION('',(0.707106781187,-0.707106781187,-2.775557561563E-16)
  );
#48 = PCURVE('',#49,#54);
#49 = CYLINDRICAL_SURFACE('',#50,25.);
[... 1341 lines cut ...]
#1110 = PROPERTY_DEFINITION('material property','material name',#432);
#1111 = PROPERTY_DEFINITION_REPRESENTATION(#1110,#1108);
#1112 = REPRESENTATION('material name',(#1113),#671);
#1113 = DESCRIPTIVE_REPRESENTATION_ITEM('Material Name 5',
  'Material description 5');
#1114 = PROPERTY_DEFINITION('material property','material name',#567);
#1115 = PROPERTY_DEFINITION_REPRESENTATION(#1114,#1112);
#1116 = REPRESENTATION('material name',(#1117),#806);
#1117 = DESCRIPTIVE_REPRESENTATION_ITEM('Material Name 6',
  'Material description 6');
#1118 = PROPERTY_DEFINITION('material property','material name',#702);
#1119 = PROPERTY_DEFINITION_REPRESENTATION(#1118,#1116);
#1120 = REPRESENTATION('material name',(#1121),#941);
#1121 = DESCRIPTIVE_REPRESENTATION_ITEM('Material Name 7',
  'Material description 7');
#1122 = PROPERTY_DEFINITION('material property','material name',#837);
#1123 = PROPERTY_DEFINITION_REPRESENTATION(#1122,#1120);
#1124 = REPRESENTATION('material name',(#1125),#1076);
#1125 = DESCRIPTIVE_REPRESENTATION_ITEM('Material Name 8',
  'Material description 8');
#1126 = PROPERTY_DEFINITION('material property','material name',#972);
#1127 = PROPERTY_DEFINITION_REPRESENTATION(#1126,#1124);
ENDSEC;
END-ISO-10303-21;
```

tpaviot, 2023-03-11:
Issue confirmed.

This is because the ```TCollection_HAsciiString``` c++ type returned by reference is not correctly wrapped.

tpaviot, 2023-03-11:
I pushed a new branch that should fix the issue (see https://github.com/tpaviot/pythonocc-core/tree/review/tollection_hasciistring).

```python

# create an handle to a document
doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))

# Get root assembly
mat_tool = XCAFDoc_DocumentTool_MaterialTool(doc.Main())

step_reader = STEPCAFControl_Reader()

status = step_reader.ReadFile(filename)
if status == IFSelect_RetDone:
    step_reader.Transfer(doc)

material_labels = TDF_LabelSequence()

mat_tool.GetMaterialLabels(material_labels)

# materials
for i in range(1, material_labels.Length() + 1):
    (
        ok,
        material_name,
        material_description,
        material_density,
        material_densname,
        material_densvaltype,
    ) = mat_tool.GetMaterial(material_labels.Value(i))

    print(f"Material name: {material_name}")
    print(f"Material description: {material_description}")
    print(f"Material density: {material_density}")
    print(f"Material densname: {material_densname}")
    print(f"Material_densvaltype: {material_densvaltype}\n")
```
produces the following output
```bash
Material name: Material Name 1
Material description: Material description 1
Material density: 7.85e-06
Material densname: Material densName 1
Material_densvaltype: 

Material name: Material Name 2
Material description: Material description 2
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 3
Material description: Material description 3
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 4
Material description: Material description 4
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 5
Material description: Material description 5
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 6
Material description: Material description 6
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 7
Material description: Material description 7
Material density: 0.0
Material densname: 
Material_densvaltype: 

Material name: Material Name 8
Material description: Material description 8
Material density: 0.0
Material densname: 
Material_densvaltype: 
```

@LuisGMM could you please update the step file so that density, density name and density val type is also available?

closed 2023-03-13 by tpaviot

LuisGMM, 2023-03-13:
It seems I cannot comment anymore :(

LuisGMM, 2023-03-13:
Great, thank you very much @tpaviot !!!

I have updated the file, you can find it [here](https://drive.google.com/file/d/1iE01movecrIrN6FY6i0c4TwC94EU-Vad/view?usp=sharing).
I have written the material attributes in the following way:
`fMaterial Densname {cnt}'`
`fMaterial Densvaltype {cnt}'`
`f'Material Description {cnt}'`
`f'Material Name {cnt}'`
`Density` is `cnt`, an integer from 1 to 8.

I have noted two things:
1. Density must be different from 0 for densname to be written, this should be the expected behavior as the information is stored in the same STEP command: `#1186 = MEASURE_REPRESENTATION_ITEM('Material Densname 8',8.,#1180);
`
2. I could not find any `Densvaltype` inside the step, nor any documentation about what that parameter is and how it is stored, maybe you have more luck tracking it. (or it is magically recovered while reading it)

PR #1184 (2023-03-13): Review/tollection hasciistring
Fix #1181