tpaviot/pythonocc-core #1194: Failing run_tests.py ERROR: test_create_mesh_datasource_from_numpy_ndarray (core_meshdatasource_unittest.TestMeshDataSource)

Daanww, 2023-04-02:
Hi,
I'm trying to compile this library from source using the instructions provided.
I managed to fully compile **release 7.7.0** but **run_tests.py** keeps failing and I'm lost in trying to solve it.
This is the error that keeps popping up:

> #11 332.4 ======================================================================
#11 332.4 ERROR: test_create_mesh_datasource_from_numpy_ndarray (core_meshdatasource_unittest.TestMeshDataSource)
#11 332.4 ----------------------------------------------------------------------
#11 332.4 Traceback (most recent call last):
#11 332.4   File "/pythonocc-core/test/core_meshdatasource_unittest.py", line 165, in test_create_mesh_datasource_from_numpy_ndarray
#11 332.4     a_data_source = MeshDS_DataSource(vertices, faces)
#11 332.4   File "/usr/lib/python3/dist-packages/OCC/Core/MeshDS.py", line 853, in __init__
#11 332.4     _MeshDS.MeshDS_DataSource_swiginit(self, _MeshDS.new_MeshDS_DataSource(*args))
#11 332.4 TypeError: Wrong number or type of arguments for overloaded function 'new_MeshDS_DataSource'.
#11 332.4   Possible C/C++ prototypes are:
#11 332.4     MeshDS_DataSource::MeshDS_DataSource(std::vector< gp_Pnt,std::allocator< gp_Pnt > >,std::vector< std::vector< int,std::allocator< int > >,std::allocator< std::vector< int,std::allocator< int > > > >)
#11 332.4     MeshDS_DataSource::MeshDS_DataSource(double *,int,int,int *,int,int)
#11 332.4     MeshDS_DataSource::MeshDS_DataSource(opencascade::handle< Poly_Triangulation > const &)
#11 332.4 
#11 332.4 
#11 332.4 ======================================================================
#11 332.4 FAIL: test_export_shape_to_svg (core_extend_dataexchange_unittest.TestExtendDataExchange)
#11 332.4 ----------------------------------------------------------------------
#11 332.4 Traceback (most recent call last):
#11 332.4   File "/pythonocc-core/test/core_extend_dataexchange_unittest.py", line 92, in test_export_shape_to_svg
#11 332.4     self.assertTrue(os.path.isfile(svg_filename))
#11 332.4 AssertionError: False is not true
#11 332.4 
#11 332.4 ----------------------------------------------------------------------
#11 332.4 Ran 130 tests in 1.867s

I'm using docker to compile it to make sure that the environment stays consistent.
Specifically I'm using **Ubuntu** as base.
This is the dockerfile that I'm using:
```
FROM ubuntu as env
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update && apt-get install -y \
  tzdata \
  python3 \
  python3-dev \
  nano \
  git \
  cmake \
  g++ \
  tcllib \
  tklib \
  tcl-dev \
  tk-dev \
  libfreetype-dev \
  libx11-dev \
  libgl1-mesa-dev \
  libfreeimage-dev \
  rapidjson-dev \
  libdraco-dev \
  swig4.0 \
  python3-pip \
  && pip install numpy \
  && pip install six
  
FROM env as clone_github
#cloning release 7.7.0
RUN git clone https://github.com/tpaviot/pythonocc-core.git --branch 7.7.0

FROM clone_github as building_occ
#copying opencascade-7.7.0 directory
COPY . .
RUN \
  cmake -S /opencascade-7.7.0 -B opencascade-build -D BUILD_RELEASE_DISABLE_EXCEPTIONS=OFF \
  && cd opencascade-build \
  && make -j`nproc`

WORKDIR /pythonocc-core

FROM building_occ as building_pythonocc
RUN \
  mkdir cmake-build \
  && cd cmake-build \
  && cmake -DOCE_INCLUDE_PATH=/opencascade-7.7.0/inc -DOCE_LIB_PATH=/opencascade-build/lin64/gcc/lib .. \
  && make -j`nproc` \
  && make install \
  #needed to be able to find the libraries
  && export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opencascade-build/lin64/gcc/lib/ \
  && cd ../test \
  && python3 run_tests.py

```

If anyone could help me that would be greatly appreciated.

tpaviot, 2023-10-12:
add
```bash
pip import svgwrite
```

SamuraiAnderson, 2023-10-22:
@tpaviot 
also need to append -DPYTHONOCC_MESHDS_NUMPY=on to solve ERROR: test_create_mesh_datasource_from_numpy_ndarray (core_meshdatasource_unittest.TestMeshDataSource)？

tpaviot, 2023-10-22:
yes