tpaviot/pythonocc-core #1195: TypeError: brepgprop_SurfaceProperties() got an unexpected keyword argument 'SkipShared'

im44pos, 2023-04-03:
As far as I understand: According to Open CASCADE Technology (https://dev.opencascade.org/doc/refman/html/class_b_rep_g_prop.html#abd91b892df8d0f6b8571deed5562ca1f) and pythonocc-core (Ref.: https://github.com/tpaviot/pythonocc-core/blob/master/src/SWIG_files/wrapper/BRepGProp.i) documentation it should be possible to use the optional SkipShared flag to compute edge length, surface area and volume.

SkipShared: bool,optional
	default value is Standard_False

I use it like this in my python code:
brepgprop_LinearProperties(shape, propl, SkipShared = True)
brepgprop_SurfaceProperties(shape, props, SkipShared = True)
brepgprop_VolumeProperties(shape, propv, SkipShared = True)

On all 3 I get a TypeError: brepgprop_...Properties() got an unexpected keyword argument 'SkipShared'

Really appreciate help on this, since the default setting "SkipShared=Standard_False" returns wrong values for overlapping geometries. Can't find any example with "SkipShared=True".

tpaviot, 2023-04-06:
issue confirmed

im44pos, 2023-04-13:
@tpaviot Thanks for confirming the issue and adding the bug label.
Since I'm relative new to programming and espessialy GitHub:
 - I noticed there's "No one assigned"
 - Should I expect this bug to be resolved? If so: when? Please clarify?
 - If not: Is there a way to create a custom function with the required functionality? Or some other solution?
I would really appreciate to have the correct computations.
Kind regards.

tpaviot, 2023-04-13:
you just have to use:
```
brepgprop_LinearProperties(shape, propl, True)
```

im44pos, 2023-04-13:
@tpaviot : Thanks for your quick response.

Now my hope is on you to confirm that you did test this.
Because then it is in my code or in the models I use to test. Either should be solvable.
If you did test: can you share the files?

I created my test models (cubes of 1 mm^3) with CadQuery-OCP, see enclosed zip file.
[step_files.zip](https://github.com/tpaviot/pythonocc-core/files/11225680/step_files.zip)
So each cube has:
 a volume of 1 mm^3, 
 6 faces with an area of 1 mm^2 = total 6 mm^2
 12 edges with a length of 1 mm = total 12 mm
For instance model "array2_2mm3_area_edge.step" contains 2 cubes assembled like they are stacked on top of eachother, 1 full face overlapping. So this assembly model should have (with SkipShared = True):
 a total volume of 2 mm^2. And it does, because no overlapping volume.
 2 times 6 mm^2 area minus 1 mm^2 = 11 mm^2 surface area (According to OCCT documentation)
 2 times 12 mm length minus 4 mm = 20 mm edge length (According to OCCT documentation)

If you didn't test, then it could turn out to really be a Bug. That I hope is soon solvable.

Looking forward to your response.
Kind regards.

im44pos, 2023-08-10:
> you just have to use:
> 
> ```
> brepgprop_LinearProperties(shape, propl, True)
> ```

@tpaviot 
I just can.t get it to work. Really appreciate your response / comments.

```propl = GProp_GProps()
brepgprop_LinearProperties(shape, propl, True)
edge_length = propl.Mass()
print(f"length \t: \t{edge_length: .3f} \t[mm]")```
Merci beaucoup

closed 2023-10-12 by tpaviot