tpaviot/pythonocc-core #1200: TypeError: in method 'HLRAlgo_EdgeIterator_Visible', argument 3 of type 'Standard_ShortReal &'

Hyakko, 2023-04-14:
hi there,
I got a problem when using HLRAlgo_EdgeIterator, code as below:

tolsta: float = 0
tolend: float = 0
It = HLRAlgo_EdgeIterator()
It.InitVisible(ed.Status())  # ed is HLRBRep_EdgeData
sta, end = It.Visible(tolsta, tolend)   # caught Error here

I tried ctypes interface, not work. I checked swig doc, it helps not much as well. Is there anyway can I pass a primitive value by reference in python?

Or is there alternative interface to get data in HLRAlgo_EdgeStatus?

tpaviot, 2023-04-14:
confirmed, there's an issue with Standard_ShortReal returned by reference in the python wrapper. Can you please provide a short code snippet that I can run on my machine, it will serve as an unit test for the fix

Hyakko, 2023-04-14:
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.HLRAlgo import HLRAlgo_EdgeIterator
from OCC.Core.HLRBRep import HLRBRep_Algo, HLRBRep_Data, HLRBRep_EdgeData, HLRBRep_FaceIterator
from OCC.Core.TopoDS import TopoDS_Shape

def InternalCompound(S: TopoDS_Shape, hlr: HLRBRep_Algo):
    """
    reproduce internalCompound of occ by python
    """
    DS: HLRBRep_Data = hlr.DataStructure()

    e1 = 1
    e2 = DS.NbEdges()
    f1 = 1
    f2 = DS.NbFaces()

    if S is not None:
        index = hlr.Index(S)
        v1, v2, e1, e2, f1, f2 = hlr.ShapeBounds(index).Bounds()
    # end if

    for ie in range(e1, e2 + 1):
        for iface in range(f1, f2 + 1):
            DrawFace(iface, DS)
        # end for
    # end for

def DrawFace(iface: int, DS: HLRBRep_Data,):
    Itf = HLRBRep_FaceIterator()
    # 初始化，搜索面上的边
    Itf.InitEdge(DS.FDataArray().ChangeValue(iface))
    while Itf.MoreEdge():
        ie: int = Itf.Edge()
        edf: HLRBRep_EdgeData = DS.EDataArray().ChangeValue(ie)
        if not edf.Used():
            DrawEdge(edf)

        Itf.NextEdge()

def DrawEdge(ed: HLRBRep_EdgeData):
    tolsta: float = 0.0
    tolend: float = 0.0

    It = HLRAlgo_EdgeIterator()
    It.InitVisible(ed.Status())
    while It.MoreVisible():
        # caught error here
        sta, end = It.Visible(tolsta, tolend)
        break
    # end while

if __name__ == '__main__':
    shapeBuilder = BRepPrimAPI_MakeBox(1, 1, 1)
    shape = shapeBuilder.Shape()
    hlr = HLRBRep_Algo()
    hlr.Add(shape)
    hlr.Projector()
    hlr.Update()
    hlr.Hide()
    print("prepare is done")

    resShape = InternalCompound(S=shape, hlr=hlr)

tpaviot, 2023-04-14:
after the fix, just use ```sta, tolsta, end, tolend = It.Visible()```

closed 2023-04-14 by tpaviot

Hyakko, 2023-04-17:
Ok, thank you

commit 554b0e19 (2023-04-14): Add transformer for Standard_ShortReal returned by reference, fix #1200