tpaviot/pythonocc-core #1204: Building on Mac M1, reprise

rckirby, 2023-04-28:
Thanks for the updates on building, I'm trying the build again.  I'm now hitting an error during the compilation process

/Users/robert_kirby/Code/firedrake/src/pythonocc-core/cmake-build/Darwin/arm64-Release-/BSplCLibPYTHON_wrap.cxx:18119:44: error: no member named 'Intervals' in 'BSplCLib'

I am using the opencascade installed with homebrew, using `cmake -DOCE_INCLUDE_PATH=/opt/homebrew/Cellar/opencascade/7.7.0/include/opencascade/ -DOCE_LIB_PATH=/opt/homebrew/Cellar/opencascade/7.7.0/lib ..` followed by `make -j8`

Am I missing a something in my include paths or homebrew dependencies?

rainman110, 2023-07-26:
Alternatively, checkout the 7.7.0 tag of pythonocc core. Then the build should work out of box.

It is important, that the pythonocc version always exactly matches the opencascade version.

vincentmarchetti, 2023-07-26:
I was able to get the OpenCascase v 7.1.1 source code from https://github.com/Open-Cascade-SAS/OCCT .
This, along with the April 19 commit of pythonocc-core (efb5cb9) and swig 4.1.1, has allowed me to build open cascade from source, and then pythococc against that installation

closed 2023-10-12 by tpaviot