tpaviot/pythonocc-core #1205: BVH of pythonocc

linbeijianbaoxia, 2023-05-06:
[image]
Is BVH package of OCCT not yet wrapped in pythonocc currently?

tpaviot, 2023-05-07:
BVH package is actually wrapped but most classes are skipped. I remember it caused a lot of issues, but I didn't get back on this topic for a while. Noone so far requested support for this class, which class method needs it?

linbeijianbaoxia, 2023-05-07:
> BVH package is actually wrapped but most classes are skipped. I remember it caused a lot of issues, but I didn't get back on this topic for a while. Noone so far requested support for this class, which class method needs it?

I need to use these classes: BVH_PrimitiveSet, BVH_Vec3d, BVH_Box, BVH_BinnedBuilder. I implemented BVH construction through OCC in C++, but it seems that I need to rewrite it in Python using pythonocc. I know that these classes also inherit from some base classes, and the encapsulation workload of these classes is huge, so I will try to build BVH from scratch in Python. Thank you very much for your selfless contribution to pythonocc.

closed 2023-05-12 by tpaviot