tpaviot/pythonocc-core #1217: init_display gives Segmentation fault (core dumped) with pythonocct-core 7.7.0 and pyside2

hongyuan1306, 2023-05-30:
With the latest pythonocc-core version 7.7.0 and pyside2, init_display() is throwing a `Segmentation fault`.

The error can be reproduced by running the following on Ubuntu 23.04:

```
conda create --name=pyoccenv python=3.9
conda activate pyoccenv
conda install -c conda-forge pythonocc-core=7.7.0 pyside2
```

and running python:

```
(pyoccenv) hong@hong-office:~/workspace/cnc$ python 
Python 3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:39:03)                    
[GCC 11.3.0] on linux                                                                      
Type "help", "copyright", "credits" or "license" for more information.                     
>>> from OCC.Display.SimpleGui import init_display                                         
>>> init_display()
qt-pyside2 backend - Qt version 5.15.8
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Segmentation fault (core dumped)
```

Running the above procedure on Win11 is OK. And also if I install `pyqt5` (using pip) instead of `pyside2`, there is also no problem.

hongyuan1306, 2023-06-01:
I find that in order to avoid the above segmentation fault, one has to call `load_backend` before **importing** `init_display`:

```
(cnc) hong@hong-office:~/workspace/cnc$ python
Python 3.10.11 | packaged by conda-forge | (main, May 10 2023, 18:58:44) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from OCC.Display.backend import load_backend
>>> load_backend()
'qt-pyside2'
>>> from OCC.Display.SimpleGui import init_display
>>> init_display()
qt-pyside2 backend - Qt version 5.15.8
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Display3d class successfully initialized.
#########################################
...
```

sjgallagher2, 2024-12-13:
Currently experiencing this issue, Debian 12, X11, KDE 5.27.5. Tried wx and pyside6 backends but both are crashing. Above workaround did not help:

```
from OCC.Display.backend import load_backend
load_backend()
Out[3]: 'pyside6'
from OCC.Display.SimpleGui import init_display
init_display()
pyside6 backend - Qt version 6.7.2
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Graphic3d_Camera created
Graphic3d_StructureManager created
Xw_Window created.
Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)
```

Environment:
```
#                                                                                                                      
# Name                    Version                   Build  Channel                                                     
_libgcc_mutex             0.1                 conda_forge    conda-forge                                               
_openmp_mutex             4.5                       2_gnu    conda-forge                                               
adwaita-icon-theme        47.0                     unix_0    conda-forge                                               
aiohappyeyeballs          2.4.3              pyhd8ed1ab_0    conda-forge                                               
aiohttp                   3.11.7          py311h2dc5d0c_0    conda-forge                                               
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge                                               
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge                                               
aom                       3.9.1                hac33072_0    conda-forge                                               
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge                                               
at-spi2-atk               2.38.0               h0630a04_3    conda-forge                                               
at-spi2-core              2.40.3               h0630a04_0    conda-forge                                               
atk-1.0                   2.38.0               h04ea711_2    conda-forge                                               
attrs                     24.2.0             pyh71513ae_0    conda-forge                                               
beautifulsoup4            4.12.3             pyha770c72_0    conda-forge                                               
blosc                     1.21.6               hef167b5_0    conda-forge                                               
brotli                    1.1.0                hb9d3cd8_2    conda-forge                                               
brotli-bin                1.1.0                hb9d3cd8_2    conda-forge                                               
bzip2                     1.0.8                h4bc722e_7    conda-forge                                               
c-ares                    1.34.3               heb4867d_0    conda-forge                                               
ca-certificates           2024.8.30            hbcca054_0    conda-forge                                               
cairo                     1.18.0               hebfffa5_3    conda-forge                                               
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge                                               
cloudpickle               3.1.0              pyhd8ed1ab_1    conda-forge
comm                      0.2.2              pyhd8ed1ab_0    conda-forge
contourpy                 1.3.1           py311hd18a35c_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
cyrus-sasl                2.1.27               h54b06d7_7    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.8.9           py311hfdbb021_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
double-conversion         3.3.0                h59595ed_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
elfutils                  0.192                h1fa0c75_0    conda-forge
epoxy                     1.5.10               h166bdaf_1    conda-forge
exceptiongroup            1.2.2              pyhd8ed1ab_0    conda-forge
executing                 2.1.0              pyhd8ed1ab_0    conda-forge
expat                     2.6.4                h5888daf_0    conda-forge
ffmpeg                    6.1.2           gpl_hdfc89ed_706    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.15.0               h7e30c49_1    conda-forge
[... 230 lines cut ...]
xorg-libxau               1.0.11               hb9d3cd8_1    conda-forge
xorg-libxcomposite        0.4.6                hb9d3cd8_2    conda-forge
xorg-libxcursor           1.2.3                hb9d3cd8_0    conda-forge
[... cut ...]
xorg-xorgproto            2024.1               hb9d3cd8_1    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yarl                      1.18.0          py311h9ecbd09_0    conda-forge
zeromq                    4.3.5                h3b0a872_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_0    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

```

sjgallagher2, 2024-12-13:
My problem seems to be related to this issue with OCCT: https://tracker.dev.opencascade.org/view.php?id=33644

tpaviot, 2024-12-13:
@sjgallagher2 which cpu/graphic card/graphic driver version?

sjgallagher2, 2024-12-13:
Will report hardware/graphics details when I get back to my computer later today. Was looking at this and it looks like it fails at [these lines](https://github.com/tpaviot/pythonocc-core/blob/74b641c425d87c8362d6b41f671e18e87a4da891/src/Visualization/Display3d.cpp#L218C2-L219C46): 
```cpp
  myV3dView->SetWindow(myWindow);
  if (!myWindow->IsMapped()) myWindow->Map();
```
So I guess a bad pointer somewhere in `myV3dView`, the `V3d_View` class in OCCT, or the `Xw_Window` class in OCCT? Something specific to Linux or my graphics (integrated intel graphics, don't have details on hand)

sjgallagher2, 2024-12-13:
Hardware: 
```
Operating System: Debian GNU/Linux 12
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 6.1.0-18-amd64 (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-8550U CPU @ 1.80GHz
Memory: 15.5 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 620
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: UX430UAR
System Version: 1.0
```

sjgallagher2, 2024-12-14:
Learning more about how the display stuff works. I'm able to create a window without issues in C++ with OCCT 7.8.0 using this code, adapted and updated from here: https://dev.opencascade.org/content/visualization-qt-what-wrong-my-code
```cpp
OCCView::OCCView(QWidget *parent)
    : QOpenGLWidget(parent)
{
    static Handle(Graphic3d_GraphicDriver) Viewer_aGraphicDriver;
    if (myGraphicDriver.IsNull()) {
        if (Viewer_aGraphicDriver.IsNull()) {
            Handle(Aspect_DisplayConnection) aDisplayConnection;
            aDisplayConnection = new Aspect_DisplayConnection(qgetenv("DISPLAY").constData());
            Viewer_aGraphicDriver = new OpenGl_GraphicDriver (aDisplayConnection);
        }
        myGraphicDriver = Handle(OpenGl_GraphicDriver)::DownCast(Viewer_aGraphicDriver);
    }

    Handle(V3d_Viewer) aViewer = new V3d_Viewer(myGraphicDriver);
    aViewer->SetDefaultBackgroundColor(Quantity_NOC_WHEAT);

    myView = aViewer->CreateView();
    AISContext = new AIS_InteractiveContext(aViewer);

    Handle(Aspect_DisplayConnection) aDispConnection = AISContext->CurrentViewer()->Driver()->GetDisplayConnection();
    //Handle(Xw_Window) hWnd = new Xw_Window(aDispConnection, "Visu3D",100,100,500,500);
    WId winid = this->winId();
    Handle(Xw_Window) hWnd = new Xw_Window(aDispConnection,winid);

    hWnd->Map();
    myView->SetWindow(hWnd);

    Handle(V3d_DirectionalLight) aLight1 = new V3d_DirectionalLight(V3d_XnegYnegZneg, Quantity_NOC_WHITE, Standard_True);
    Handle(V3d_AmbientLight) aLight2 = new V3d_AmbientLight(Quantity_NOC_WHITE);

    myView->ZBufferTriedronSetup();
    myView->TriedronDisplay(Aspect_TOTP_LEFT_LOWER, Quantity_NOC_BLACK, 0.1, V3d_ZBUFFER);
    aViewer->AddLight(aLight1);
    aViewer->AddLight(aLight2);
    aViewer->SetLightOn();
    aViewer->SetDefaultBackgroundColor(Quantity_NOC_GRAY);

    Handle (AIS_Shape) aBox = new AIS_Shape (BRepPrimAPI_MakeBox(5, 5, 5).Solid());
    AISContext->SetDisplayMode (AIS_Shaded,Standard_True);
    AISContext->Display(aBox,Standard_True);
    myView->FitAll();
}
```

So there's something wrong with the way the window handles are being managed, maybe what is explained in the post I linked. 

> As jiping xin pointed out, the program crashes on the line "aWindow = new Xw_Window(...)", the problem being that aDisplayConnection points to invalid data.
> 
> This is apparently expected on Windows and Mac, but for Linux it has to be set to the current display. The solution is therefore to add the following line just after the definition of "aDisplayConnection":
> 
```cpp
Handle(Aspect_DisplayConnection) aDisplayConnection;
aDisplayConnection = new Aspect_DisplayConnection(qgetenv("DISPLAY").constData());
```

In my case the program is crashing when calling `SetWindow()` or when mapping, so it seems different than the above, but could be in a similar category?

tpaviot, 2024-12-15:
I'm not sure it is related to the ```Aspect_DisplayConnection```. Whenever I encountered this type of error, it always had to do with the GUI manager window id. Do you use x11 or wayland?

sjgallagher2, 2024-12-15:
X11, see specs above

Coder-ZZY, 2025-01-15:
I also meet this problem, how can I solve it?
`
>>> from OCC.Display.backend import load_backend
>>>  load_backend()
  File "<stdin>", line 1
    load_backend()
IndentationError: unexpected indent
>>> load_backend()
'qt-pyqt5'
>>>  from OCC.Display.SimpleGui import init_display
  File "<stdin>", line 1
    from OCC.Display.SimpleGui import init_display
IndentationError: unexpected indent
>>> from OCC.Display.SimpleGui import init_display
>>> init_display
<function init_display at 0x7ea291cb90d0>
>>> init_display()
qt-pyqt5 backend - Qt version 5.15.6
Segmentation fault (core dumped)`