tpaviot/pythonocc-core #1230: Can't work on wayland

ovo-Tim, 2023-07-10:
When I use `qtViewer3d` on wayland:
```
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x2
  Serial number of failed request:  29
  Current serial number in output stream:  30
```
That's because `QWidget.winId()` can't return right winID on Wayland, more information: https://forum.qt.io/post/764182
So, is there any other way to display on qt？

rainman110, 2023-07-12:
Though I don't have a solution for you, it might be a good idea to look into the occt / qopenglwidget example: https://github.com/gkv311/occt-samples-qopenglwidget

I think we might be able to replicate the code with python and pythonocc.

This code basically move the OpenGl context management to qt. OCCT simply renderes into the context created by qt. This does not need to call `winId`, since the OpenGl context is already created.

ovo-Tim, 2023-07-19:
I have a idea, it's not final solution, but it can work.
```
if platform.system() == 'Linux':
    if os.popen('echo $XDG_SESSION_TYPE').read() == 'wayland\n':
        os.environ['QT_QPA_PLATFORM'] = 'xcb'
```

tpaviot, 2023-10-03:
I have been trying to move up to Qt6 (PyQt6 and PySide6), and face, of course, the same issue. The 'QT_QPA_PLATFORM' trick does not work on my linux machine using PyQt6 installed from `pip`.

A pointer:
https://github.com/inviwo/inviwo/blob/aa744757d540dfb9a2b54dba646a878fc70c1478/apps/inviwopyapp/inviwo.py

tpaviot, 2023-10-12:
#1255 fixes the issue

closed 2023-10-12 by tpaviot

ovo-Tim, 2023-10-12:
> #1255 fixes the issue
I don't think that is the final solution. We should native support for the wayland instad of using xcb.
I think we should rewrite [Display3d.cpp](https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Display3d.cpp), but converting between different languages makes that difficult.

tpaviot, 2023-10-13:
@ovo-Tim You're right, a specific linux tweak is required for this specific case, any contribution will be welcome.

ovo-Tim, 2023-10-13:
> Though I don't have a solution for you, it might be a good idea to look into the occt / qopenglwidget example: https://github.com/gkv311/occt-samples-qopenglwidget
>
> I think we might be able to [...]

@tpaviot I have no idea about how to realize it. But I think @rainman110 's advice is useful.
 
@rainman110 Hi, have you made any progress? 😘

tpaviot, 2023-10-13:
I guess the key is here: https://github.com/gkv311/occt-samples-qopenglwidget/blob/master/occt-qopenglwidget/OcctQtViewer.cpp#L375

ovo-Tim, 2023-10-14:
OK, I have posted about this problem.
https://dev.opencascade.org/content/it-possible-native-support-wayland#comment-25077 https://tracker.dev.opencascade.org/view.php?id=33505

ovo-Tim, 2023-10-15:
It shouldn't be closed.