tpaviot/pythonocc-core #1231: Missing Qt-Platform-Plugin "wayland" - pySide / pyQt / visualization

martinjan10, 2023-07-12:
Hey together, I'm getting started with pythonOCC for my thesis and I'm already struggeling at the installation-part :D but motivation is high.

So: I managed to install pythonOCC (7.7.0) inside a venv with python 3.9 via Anaconda (took me longer than I want to admit...) on my Ubuntu 22.04 machine.

Anaconda installation: [https://problemsolvingwithpython.com/01-Orientation/01.05-Installing-Ana...](https://problemsolvingwithpython.com/01-Orientation/01.05-Installing-Anaconda-on-Linux/)
Adding it to PATH and adding conda-forge as a channel (otherwise pythonOCC was not found):

`cd source .bashrc 
# https://conda-forge.org/ 
conda config --add channels conda-forge 
conda config --set channel_priority strict `

Installing pythonOCC inside a venv: https://github.com/tpaviot/pythonocc-core  
`conda create --name=pyoccenv python=3.9 
source activate pyoccenv 
conda install -c conda-forge pythonocc-core=7.7.0`

After all that I just want to run an example-code to test if everything works:
`from OCC.Display.SimpleGui import init_display

from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()

my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()

display.DisplayShape(my_box, update=True)

start_display()`

I had a Error message, that there is no visualization backend (or similar, sadly I don't remember the exact wording). Because of that and the fact that I read about similar topics before I installed pySide2 via conda inside the venv (conda install -c conda-forge pyside2). After that the error while executing the output changed to: 

`qt - pyside2 backend - Qt version ..
qt-qpa-plugin: Could not find the Qt platform plugin "wayland" in ""
###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGL_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d View created`

but nothing is visible...

I tried installing the following things to solve the problem: 

`# wxpython
sudo apt update

sudo apt install libgtk-3-dev

sudo pip install wxpython # dauert seeeehr lange

# pyqt
conda install pyqt

# qtwayland5
sudo apt install qtwayland5

export QT_QPA_PLATFORM=wayland`

But nothing worked till now. only the "pyside2" before the "backend" changed to "pyqt5". I would love to hear your thoughts regarding this topic and if someone stumbled upon this error before.

Thaaanks and greetings

rainman110, 2023-07-12:
This issue might be related to #1230 .

tpaviot, 2023-10-12:
Support for recent Qt6 gui managers has been recently added. For your specific problem, try:

```bash
export QT_QPA_PLATFORM=xcb
```

closed 2023-10-12 by tpaviot