tpaviot/pythonocc-core #1232: BRepAlgoAPI_Cut cutting order

giovannicono, 2023-07-13:
Hi,
I have two verified solid objects in input to the BRepAlgoAPI_Cut tool, a blade and a slice. If the blade is not vertical swirled I get a perfect cut with BRepAlgoAPI_Cut(bladeIGES_Slice, bladeIGES_Bspline), but if it's swirled I get a cutted blade on tip and hub. I can not figure out where is the problem, the two blade files are exactly the same for dimension etc, this is the only parameter that change. 
I attach the examples of the input objects and the two outputs.
[image]

Tanneguydv, 2023-07-13:
Hello,
could you provide example .stp files too please?

giovannicono, 2023-07-17:
Sorry, I sent it by mail but the file was too large. I upload it here.
[Example.zip](https://github.com/tpaviot/pythonocc-core/files/12067370/Example.zip)

Tanneguydv, 2023-07-17:
It's ok for me, for both shapes :
Straight blade :
[image]

Swirled blade :
[image]

Be sure to respect the correct order of the method parameters : 
`BRepAlgoAPI_Cut(basis, cutter)`

basis = "argument" and cutter="tool" in this description -->

[https://dev.opencascade.org/doc/refman/html/class_b_rep_algo_a_p_i___cut.html](https://dev.opencascade.org/doc/refman/html/class_b_rep_algo_a_p_i___cut.html)

giovannicono, 2023-07-17:
The order is the same for both cutting operation because the function is the same. The blades and slices are always generated with PythonOCC, then used for the cutting operation. I generate the 4 faces of the blade and then sew it together, returning the .Shape() to pass to the boolean subtractor. Maybe is there a way to return a more stable object?