tpaviot/pythonocc-core #1233: OCC.Core.TopExp - Vertices Doubled (?)

gtagliani123, 2023-07-17:
Hi guys! just wondering if this is a bug of pythoncore-occ or freecad is doing weird stuff. 

So, I created a simple Skecth in freecad to test how to read the open cascade brp files that freecad generates with pythonocc-core and I encountered that using "OCC.Core.TopExp - Init(shape, TopAbs_VERTEX)" I get 6 vertices instead of 3. 

so... my code might be wrong... could this be a bug of pythonocc-core ? or is it just the freecad brp files that are weird ?

I upload my code and the BRP file, 

Output of the execution: 
```
Vertices: 6
divide by 2? 
Edges: 3
Faces: 0
```

##########################

Image of sketch: 
(clearly has 3 vertices)
[image]

Code:
 
```
from OCC.Core.BRepTools import breptools_Read
from OCC.Core.TopoDS import TopoDS_Shape
from OCC.Core.BRep import BRep_Builder
from OCC.Core.TopAbs import TopAbs_VERTEX, TopAbs_EDGE, TopAbs_FACE
from OCC.Core.TopExp import TopExp_Explorer

def read_brep_file(filename):
    shape = TopoDS_Shape()
    builder = BRep_Builder()
    breptools_Read(shape, filename, builder)
    return shape

def count_topology(shape):
    count_dict = {'Vertices': 0, 'Edges': 0, 'Faces': 0}
    explorer = TopExp_Explorer()

    explorer.Init(shape, TopAbs_VERTEX)
    while explorer.More():
        count_dict['Vertices'] += 1
        explorer.Next()

    explorer.Init(shape, TopAbs_EDGE)
    while explorer.More():
        count_dict['Edges'] += 1
        explorer.Next()

    explorer.Init(shape, TopAbs_FACE)
    while explorer.More():
        count_dict['Faces'] += 1
        explorer.Next()

    return count_dict

filename = './unziped/PartShape1.brp'  ### file of the freecad sketch
shape = read_brep_file(filename)
count_dict = count_topology(shape)

for key, value in count_dict.items():
    print(f"{key}: {value}")
    if key=="Vertices":
        print("divide by 2? ")

```

BRP FILE THAT reads

```

CASCADE Topology V1, (c) Matra-Datavision
Locations 1
1
1.000000000000000 0.000000000000000 0.000000000000000 0.000000000000000 
0.000000000000000 1.000000000000000 0.000000000000000 0.000000000000000 
0.000000000000000 0.000000000000000 1.000000000000000 0.000000000000000 
Curve2ds 0
Curves 3
1 -82.39437100000000669 10.38732500000000059 0.00000000000000000 0.94265518296116502 -0.33376819206816638 0.00000000000000000 
1 -42.42957700000000187 36.79577599999999649 0.00000000000000000 -0.83430516396390397 -0.55130290529178549 0.00000000000000000 
1 -42.42957700000000187 36.79577599999999649 0.00000000000000000 0.92985169854002936 -0.36793453048364799 0.00000000000000000 
Polygon3D 0
PolygonOnTriangulations 0
Surfaces 0
Triangulations 0

TShapes 8
Ve
0.000000100000000
-82.394371000000007 10.387325000000001 0.000000000000000
0 0

0101101
*
Ve
0.000000100000000
73.239432999999991 -44.718311000000000 0.000000000000000
0 0

0101101
*
Ed
 0.000000100000000 1 1 0
1  1 0 0.000000000000000 165.101520485048582
0

0101000
+8 0 -7 0 *
Ve
0.000000100000000
-42.429577000000002 36.795775999999996 0.000000000000000
0 0

0101101
*
Ed
 0.000000100000000 1 1 0
1  2 0 0.000000000000000 47.901889771509403
0

0101000
+5 0 -8 0 *
Ve
0.000000100000000
112.852112000000005 -24.647887999999995 0.000000000000000
0 0

0101101
*
Ed
 0.000000100000000 1 1 0
1  3 0 0.000000000000000 166.996187933849598
0

0101000
+5 0 -3 0 *
Wi

1101000
-6 0 -4 0 +2 0 *

+1 1 

```

jf---, 2023-07-24:
Every edge has 2 unique vertices these aren't shared ( as is the case for meshes ) so 3 edges x 2 vertices per edge, as expected

gtagliani123, 2023-07-24:
Oh!!!! lol! ok! Ty so much, closed! :D

closed 2023-07-24 by gtagliani123