tpaviot/pythonocc-core #1236: How to integrate pythonocc-core into a Blender addon

DaJansenGit, 2023-07-25:
This is not so much an issue as a question, but I hope to find an answer in this community.

We are working on a research project and want to create a Blender addon built on top of BlenderBIM. Our python library uses IfcOpenShell and PythonOCC. The problem is that current versions of PythonOCC can only be installed via Anaconda, and we can't manage to integrate PythonOCC with Blender's built-in Python interpreter.
We have tried both:
* copy a complete anaconda installation of PythonOCC including all dependencies into the BlenderBIM interpreter.
* Build PythonOCC ourselves, which failed (unfortunately the instructions in the readme.md seem to be out of date).

Does anyone have any ideas on how to integrate PythonOCC correctly, or can direct me to the right place to look for more information? Any help is appreciated.

RomainGuimbal, 2023-07-28:
Hi,
Do you know about this addon ? [https://blenderartists.org/t/step-import/1203804](https://blenderartists.org/t/step-import/1203804)
It contains a build of Python OCC that works great. Like you I tried to copy my anaconda build and it failed too, so I directly used the build of the addon as a temporary solution. I sadly don't have a better solution yet, but the issue is might be linked to Visual Studio Runtime (a specific version of it is required for the addon and I have the same error as this post https://blenderartists.org/t/step-import/1203804/29 and it was the issue)

Also you can directly ask Ambi, it might be the easiest way.

rainman110, 2023-07-28:
@DaJansenGit i recommend to build pythonocc by yourself. You said, you failed building it .

Can you please give more detail, what build issues you encountered?

tpaviot, 2023-07-30:
@DaJansenGit, I recommend using conda. Building pythonocc itself is not an issue; I'll double-check the build instructions to see if they might be outdated. However, the build process has not changed for a while. The real problem lies in managing binary dependencies. Unfortunately, the opencascade team does not maintain any PPA or similar package repositories. If you want your Blender addon to support Linux, Windows, and macOS, you'll have to handle all the packaging tasks by yourself, and it can be quite time-consuming and cumbersome. That's why I made the switch to conda a few years ago. It allowed me to get rid of those tedious compiling tasks and streamline the packaging process significantly. It almost feels like a full-time job just to manage these tasks, so conda was a lifesaver in that regard.

DaJansenGit, 2023-07-31:
Thanks for your answers!

@RomainGuimbal: we came across STEPper which looks promising. But as far as I understand, everyone using the Addon we want to develop would need to buy STEPper as well and as we are working in a founded research project we can't rely/don't want to rely on commercial products in our toolchain. 

@rainman110 I wasn't building it on my own, it was a software developer at our department. I will forward him here to share the logs with you.

@tpaviot I totally understand your step towards conda. As mentioned before, I myself can't share the logs, but I hope my colleague can. 

We are not as well in contact with the guys from Blender BIM because that an integration of PythonOCC before by using anaconda. If we make any progress, we will share it here as well.

RomainGuimbal, 2023-08-02:
> Thanks for your answers!
>
> @RomainGuimbal: we came across STEPper which looks promising. But as far as I understand, everyone using the Addon we want to develop would need to buy STEPper as well and [...]

Actually, as it is a Blender plugin it can't be a commercial product, the license inside is open, it is only tolerated to sell it on commercial platform.