tpaviot/pythonocc-core #1237: Export Geom_BSplineSurface to STEP

RomainGuimbal, 2023-07-28:
Hi,
I am new to occt and I struggle to export some `Geom_BSplineSurface` to STEP files. 
I naïvely tried `write_step_file(my_geom, "C:/my_path")` without success and as far as I understand the geom objects are not "a shape" and I am a bit confused over how to make the conversion. I found many packages that could be the one, but without explanations of how to use them. I appreciate any help :)

tpaviot, 2023-07-30:
```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeTorus
from OCC.Extend.DataExchange import write_step_file

a_topods_shape = BRepPrimAPI_MakeTorus(200, 50).Shape()
write_step_file(a_topods_shape, "torus.stp", application_protocol="AP203")
```

RomainGuimbal, 2023-08-17:
> ```python
> from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeTorus
> from OCC.Extend.DataExchange import write_step_file
> 
> a_topods_shape = BRepPrimAPI_MakeTorus(200, 50).Shape()
> write_step_file(a_topods_shape, "torus.stp", application_protocol="AP203")
> ```

Yes, this I know but how can it be used for bezier surfaces?

RomainGuimbal, 2023-08-21:
I think I wasn't very clear. I have a Geom_BSplineSurface defined (and it seems okay since I can display it).
But when trying to export to step it must be the wrong type because I get either :

` 'Geom_BSplineSurface' object has no attribute 'IsNull'` with :
`write_step_file(My_geom, "test.stp", application_protocol="AP203")`

Or ` 'Geom_BSplineSurface' object has no attribute 'shape'`
with `write_step_file(My_geom.shape(), "test.stp", application_protocol="AP203")`

RomainGuimbal, 2023-08-23:
Solution found here [https://github.com/tpaviot/pythonocc-core/issues/735](https://github.com/tpaviot/pythonocc-core/issues/735)

```python
from OCC.Extend.ShapeFactory import make_face

face = BRepBuilderAPI_MakeFace(My_geom, 1e-6).Face()
write_step_file(face, "test.stp", application_protocol="AP203")
```

closed 2023-08-23 by RomainGuimbal