tpaviot/pythonocc-core #1244: how to read step faster

codazq, 2023-09-12:
an example at https://github.com/dblanding/kodacad @dblanding 

i try to open a 20MB stp file, it cost about 50s and using threading cannot shorter that time.

Are there any known algorithms that can improve this reading speed?

Thank you very much for your great contributions

codazq, 2023-09-12:
[stepanalyzer.txt](https://github.com/tpaviot/pythonocc-core/files/12588562/stepanalyzer.txt)
and his step read code

dblanding, 2023-09-12:
I'm sorry cadazq, but I don't have a way to make the read proceed more quickly. Have you tried opening your step file with Cad_Assistant? One thing you might try is first loading your step file into Cad_Assistant, then saving it back again (either in AP-214 or AP-203), then try loading the newly saved file to see if that makes it load more quickly.

codazq, 2023-09-12:
> I'm sorry cadazq, but I don't have a way to make the read proceed more quickly. Have you tried opening your step file with Cad_Assistant? One thing you might try is first loading your step file into [...]

very thanks for reply. CAD Assistant work really fast. it count 16s to open a same stp. i just wondering how to make it happen.:)

codazq, 2023-09-12:
> I'm sorry cadazq, but I don't have a way to make the read proceed more quickly. Have you tried opening your step file with Cad_Assistant? One thing you might try is first loading your step file into [...]

really cool. it cost 28s after i loading & saving. more curious...

tpaviot, 2023-09-22:
CAD Assistant is built on opencascade as well. I think there should be some kind of multitask optimizations at CAD Assitant.

dpasukhi, 2024-09-25:
CAD Assistant has no optimization for ImportStep file.
The difference are ShapeHealing parameters. In python case that parameters can be missed.