tpaviot/pythonocc-core #1245: MakePipe Troubleshooting

nsmela, 2023-09-12:
Good day,

I'm trying to create a function where I can input a list of three points and convert it into a solid pipe to subtract from another solid.

```
# source: https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_pipe.py
def execute(points):
    # Creation of points for the spine
    length = len(points)
    array  = TColgp_Array1OfPnt(0, length)
    for index in range(length):
        point = points[index]
        print(point)
        array.SetValue(index, gp_Pnt(point[0], point[1], point[2]))
        
    # Creation of a Bezier Curve as the spine
    bz_curv = Geom_BezierCurve(array)
    bz_curv_edge = BRepBuilderAPI_MakeEdge(bz_curv).Edge()
    bz_curv_wire = BRepBuilderAPI_MakeWire(bz_curv_edge).Wire()

    # Creation of profile to sweep along the spine
    radius = 1.60
    circle = gp_Circ(gp_ZOX(), radius)
    circle_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
    circle_wire = BRepBuilderAPI_MakeWire(circle_edge).Wire()

    # Creation of the law to dictate the evolution of the profile
    brep1 = BRepOffsetAPI_MakePipeShell(bz_curv_wire)
    law_f = Law_Linear()
    law_f.Set(0, 1, 1, 1)
    brep1.SetLaw(circle_wire, law_f, True, True)
    brep1.Build()
    brep1.MakeSolid()
    return brep1.Shape()
```

The issue is that there seems to be a point added to the curve at 0, 0, 0 and it affect any point array I add.

nsmela, 2023-09-13:
Made some corrections, but instead of grouping at 0,0,0, the tops of the paths are squiggly

Example Data:
```
[12.734671415451, -4.014806351862397, 222.7628290443381]
[13.117366356742401, -5.5331802141921, 218.01434661348654]
[13.500061298034101, -7.051554076521899, 213.265864182635]
[13.8827562393258, -8.569927938851599, 208.51738175178352]
[14.265451180617502, -10.088301801181299, 203.7688993209319]
[14.648146121909202, -11.606675663511098, 199.02041689008]
[15.030841063200901, -13.125049525840797, 194.27193445922902]
[15.413536004492602, -14.6434233881706, 189.523452028377]
[15.796230945784302, -16.1617972505003, 184.774969597526]
[16.178925887076, -17.68017111283, 180.026487166674]
```

New Code:
```
def generate_3(points):
    # ref: https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_make_pipe_shell.py
    # convert points into a TColgp_Array1OfPnt into a bezier curve into a pipe
    from OCC.Core.TColgp import TColgp_Array1OfPnt
    from OCC.Core.Geom import Geom_BezierCurve, Geom_BSplineCurve, Geom_Circle
    from OCC.Core.GeomFill import GeomFill_Pipe
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeShell
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipeShell
    from OCC.Core.Law import Law_Linear
    
    length = len(points)
    poles = TColgp_Array1OfPnt(0, length)
    for i in range(length):
        point = gp_Pnt(points[i][0], points[i][1], points[i][2])
        poles.SetValue(i, point)

    bz_curve = Geom_BezierCurve(poles)
    bz_curve_edge = BRepBuilderAPI_MakeEdge(bz_curve).Edge()
    bz_curve_wire = BRepBuilderAPI_MakeWire(bz_curve_edge).Wire()
    
    # profile
    origin = gp_Pnt(points[0][0], points[0][1], -1.0)
    direction = gp_Dir(0, 0, 1)
    vector = gp_Ax2(origin, direction)
    radius = 1.60
    circle = gp_Circ(vector, radius)
    circle_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
    profile = BRepBuilderAPI_MakeWire(circle_edge).Wire()
    
    # pipe
    pipeBuilder = BRepOffsetAPI_MakePipeShell(bz_curve_wire)
    law_f = Law_Linear()
    law_f.Set(0, 0.5, 1, 1)
    pipeBuilder.SetLaw(profile, law_f, False, True)
    return pipeBuilder.Shape()
```

Result:
[image]

tnakaicode, 2023-09-14:
I got the same result. bz_curve_wire is in the right shape?

nsmela, 2023-09-14:
Not sure how to check. I managed to brute-force the problem by making shapes instead and fusing them:
```
def generate_4(points):
    # ref: https://stackoverflow.com/questions/47163841/pythonocc-opencascade-create-pipe-along-straight-lines-through-points-profile
    # using the cylinders for the tube and spheres for the connections
    from OCC.Core.gp import gp_Pnt, gp_Circ, gp_Ax2, gp_Dir
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeFace
    from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeSphere
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipe
    from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
    
    # convert [x, y, z] to gp_Pnt(x, y, z)
    array = []
    for point in points:
        array.append(gp_Pnt(point[0], point[1], point[2]))
    array.append(gp_Pnt(points[-1][0], points[-1][1], -1.0))    

    # generate cylinders
    radius = 1.60
    pipes = []
    for i in range(len(array) - 1):
        edge = BRepBuilderAPI_MakeEdge(array[i], array[i + 1]).Edge()
        makeWire = BRepBuilderAPI_MakeWire(edge)
        makeWire.Build()
        wire = makeWire.Wire()
        
        direction = gp_Dir(
            array[i + 1].X() - array[i].X(),
            array[i + 1].Y() - array[i].Y(),
            array[i + 1].Z() - array[i].Z()
        )
        circle = gp_Circ(gp_Ax2(array[i], direction), radius)
        profile_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
        profile_wire = BRepBuilderAPI_MakeWire(profile_edge).Wire()
        profile_face = BRepBuilderAPI_MakeFace(profile_wire).Face()
        pipes.append(BRepOffsetAPI_MakePipe(wire, profile_face).Shape())
    
    # fuse pipes with a sphere at each joining point
    pipe = pipes[0]
    for i in range(len(array) - 1):
        sphere = BRepPrimAPI_MakeSphere(array[i], radius).Shape()
        pipe = BRepAlgoAPI_Fuse(pipe, sphere).Shape()
        pipe = BRepAlgoAPI_Fuse(pipe, pipes[i]).Shape()
    
    return pipe
```

I tried doing the wires and filleg example in the referenced link above by doing this:
```
def generate_3(points):
    # ref: https://stackoverflow.com/questions/47163841/pythonocc-opencascade-create-pipe-along-straight-lines-through-points-profile
    # generate the pipe as sections and fillet the joins
    # fails because the gp_Pln is invalid on thef illet side
    from OCC.Core.gp import gp_Pnt, gp_Dir, gp_Circ, gp_Pln, gp_Ax2
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeFace
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipe
    from OCC.Core.TColgp import TColgp_Array1OfPnt
    
    # the points
    length = len(points)
    poles = TColgp_Array1OfPnt(0, length)
    for i in range(length):
        point = gp_Pnt(points[i][0], points[i][1], points[i][2])
        poles.SetValue(i, point)

    # the edges
    edges = []
    for i in range(len(poles) - 1):
        edge = BRepBuilderAPI_MakeEdge(poles[i], poles[i + 1]).Edge()
        edges.append(edge)

    # inbetween
    fillet_radius = 0.3
    fillets = []
    for i in range(len(poles) - 2):
        # if we send the referenced edges, they'll be nulled as the fillet algo continues
        p1 = poles[i]
        p2 = poles[i + 1]
        p3 = poles[i + 2]
        e1 = BRepBuilderAPI_MakeEdge(p1, p2).Edge()
        e2 = BRepBuilderAPI_MakeEdge(p2, p3).Edge()
        fillet = filletEdges(e1, e2, fillet_radius)
        fillets.append(fillet)

    # the wire
    makeWire = BRepBuilderAPI_MakeWire()
    for i in range(len(edges)):
        makeWire.Add(edges[i].Edge())
        makeWire.Add(fillets[i])

    makeWire.Build()
    wire = makeWire.Wire()
    # the pipe
    radius = 1.60
    dir = gp_Dir(0,1,0)
    circle = gp_Circ(gp_Ax2(array[0],dir), radius)
    profile_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
    profile_wire = BRepBuilderAPI_MakeWire(profile_edge).Wire()
    profile_face = BRepBuilderAPI_MakeFace(profile_wire).Face()
    return BRepOffsetAPI_MakePipe(wire, profile_face).Shape()

def filletEdges(ed1, ed2, radius, plane):
    from OCC.Core.ChFi2d import ChFi2d_AnaFilletAlgo
    from OCC.Core.gp import gp_Pln
    f = ChFi2d_AnaFilletAlgo()
    f.Init(ed1, ed2, gp_Pln())
    f.Perform(radius)
    result = f.Result(ed1, ed2)
    return result
```
But it failed. Produced null fillets and somehow turned the supplied edges also into nulls. I suspect it's due to not having the correctly supplied gp_Pln, but I'm just a beginner.

tnakaicode, 2023-09-14:
The ```filletEdge``` in ```generate_3``` is returning null.
ChFi2d_AnaFilletAlgo requires Edges like the green and blue in the attached image, but in your code there is no angle between the Edges.
Maybe this sample would be helpful?　https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_2d_fillet.py

[image]

tnakaicode, 2023-09-14:
How about
```
from OCC.Core.gp import gp_Pnt, gp_Pln, gp_Trsf
from OCC.Core.ChFi2d import ChFi2d_AnaFilletAlgo
from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_ThruSections
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.Extend.ShapeFactory import make_wire
from OCC.Core.TopLoc import TopLoc_Location

from OCC.Display.SimpleGui import init_display

if __name__ == "__main__":
    display, start_display, add_menu, add_function_to_menu = init_display()

    dats = [
        # p1(x,y,z), p2(x,y,z), p3(x,y,z), radii, z
        [[0, 0], [5, 5], [-5, 5], 0.3, 0],
        [[0, 0], [5, 5], [-5, 5], 1.0, 1],
        [[0, 0], [5.1, 5], [-5, 5], 2.0, 2],
        [[0, 0], [5.0, 4.9], [-5, 5], 1.5, 3.5],
        [[0, 0], [5.1, 4.9], [-5.1, 5.1], 1.1, 4],
    ]

    api = BRepOffsetAPI_ThruSections()
    for data in dats:
        radius, z = data[3], data[4]
        # Defining the points
        p1 = gp_Pnt(*data[0], 0)
        p2 = gp_Pnt(*data[1], 0)
        p3 = gp_Pnt(*data[2], 0)

        # Making the edges
        ed1 = BRepBuilderAPI_MakeEdge(p3, p2).Edge()
        ed2 = BRepBuilderAPI_MakeEdge(p2, p1).Edge()

        # Making the 2dFillet
        f = ChFi2d_AnaFilletAlgo()
        f.Init(ed1, ed2, gp_Pln())
        f.Perform(radius)
        fillet2d = f.Result(ed1, ed2)

        trf = gp_Trsf()
        trf.SetTranslation(gp_Pnt(0, 0, 0), gp_Pnt(0, 0, z))

        # Create and display a wire
        w = make_wire([ed1, fillet2d, ed2])
        w.Location(TopLoc_Location(trf))
        display.DisplayShape(w)
        # display.DisplayMessage(p1, "p1")
        # display.DisplayMessage(p2, "p2")
        # display.DisplayMessage(p3, "p3")
        # display.DisplayShape(ed1)
        # display.DisplayShape(ed2)

        api.AddWire(w)
    api.Build()
    display.DisplayShape(api.Shape())

    display.FitAll()
    start_display()
```

nsmela, 2023-09-14:
Sorry, didn't mean to get fixated on the fillets. I'm not sure how to implement what you've posted with the list of points formatted that way.

Here's a better reference for my issue:
```

def generate(points): 
    # ref: https://neweopencascade.wordpress.com/2014/03/25/build-simple-pipe/
    # convert points into a TColgp_Array1OfPnt into a bezier curve
    from OCC.Core.gp import gp_Pnt, gp_Dir, gp_Ax2
    from OCC.Core.TColgp import TColgp_Array1OfPnt
    from OCC.Core.Geom import Geom_BezierCurve, Geom_BSplineCurve, Geom_Circle
    from OCC.Core.GeomFill import GeomFill_Pipe
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeShell
    length = len(points)
    poles = TColgp_Array1OfPnt(0, length)
    for i in range(length):
        point = gp_Pnt(points[i][0], points[i][1], points[i][2])
        poles.SetValue(i, point)

    path = Geom_BezierCurve(poles)
    
    # profile
    origin = gp_Pnt(points[0][0], points[0][1], -1.0)
    direction = gp_Dir(0, 0, 1)
    vector = gp_Ax2(origin, direction)
    radius = 1.60
    profile = Geom_Circle(vector, radius)
    
    # pipe
    pipeBuilder = GeomFill_Pipe(path, profile)
    pipeBuilder.Perform()
    surface = pipeBuilder.Surface()
    shell = BRepBuilderAPI_MakeShell(surface).Shell()
    return shell

def generate_2(points):
    # ref: https://stackoverflow.com/questions/47163841/pythonocc-opencascade-create-pipe-along-straight-lines-through-points-profile
    # using the cylinders for the tube and spheres for the connections
    from OCC.Core.gp import gp_Pnt, gp_Circ, gp_Ax2, gp_Dir
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeFace
    from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeSphere
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipe
    from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
    
    # convert [x, y, z] to gp_Pnt(x, y, z)
    array = []
    for point in points:
        array.append(gp_Pnt(point[0], point[1], point[2]))
    array.append(gp_Pnt(points[-1][0], points[-1][1], -1.0))    

    # generate cylinders
    radius = 1.60
    pipes = []
    for i in range(len(array) - 1):
        edge = BRepBuilderAPI_MakeEdge(array[i], array[i + 1]).Edge()
        makeWire = BRepBuilderAPI_MakeWire(edge)
        makeWire.Build()
        wire = makeWire.Wire()
        
        direction = gp_Dir(
            array[i + 1].X() - array[i].X(),
            array[i + 1].Y() - array[i].Y(),
            array[i + 1].Z() - array[i].Z()
        )
        circle = gp_Circ(gp_Ax2(array[i], direction), radius)
        profile_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
        profile_wire = BRepBuilderAPI_MakeWire(profile_edge).Wire()
        profile_face = BRepBuilderAPI_MakeFace(profile_wire).Face()
        pipes.append(BRepOffsetAPI_MakePipe(wire, profile_face).Shape())
    
    # fuse pipes with a sphere at each joining point
    pipe = pipes[0]
    for i in range(len(array) - 1):
        sphere = BRepPrimAPI_MakeSphere(array[i], radius).Shape()
        pipe = BRepAlgoAPI_Fuse(pipe, sphere).Shape()
        pipe = BRepAlgoAPI_Fuse(pipe, pipes[i]).Shape()
    
    return pipe

data = [
    [-33.708023525841, -2.496432489532598, 206.2113114751896],
    [-33.325328584549, -4.014806351862397, 201.4628290443381],
    [-32.9426336432576, -5.5331802141921, 196.71434661348653],
    [-32.559938701965905, -7.051554076521899, 191.965864182635],
    [-32.1772437606742, -8.569927938851599, 187.2173817517835],
    [-31.7945488193825, -10.088301801181299, 182.46889932093188],
    [-31.411853878090803, -11.606675663511098, 177.72041689008],
    [-31.0291589367991, -13.125049525840797, 172.971934459229],
    [-30.6464639955074, -14.6434233881706, 168.22345202837698],
    [-30.2637690542157, -16.1617972505003, 163.47496959752598],
    [-29.881074112924, -17.68017111283, 158.726487166674]
]

if __name__ == "__main__":
    from OCC.Core.Graphic3d import Graphic3d_NOM_TRANSPARENT
    from OCC.Display.SimpleGui import init_display    
    display, start_display, add_menu, add_function_to_menu = init_display()
    
    bezier_pipe = generate(data)
    display.DisplayShape(bezier_pipe, update = False)
    
    joined_pipe = generate_2(data)
    display.DisplayShape(joined_pipe, update = True, material = Graphic3d_NOM_TRANSPARENT)
    
    start_display()
    
```

This script produces:
[image]

The transparent channel is a literal representation of the same data points converted into a pipe, just doesn't have the same curve I'd like. The line is the bezier curve, which is pretty crazy. The curve is no-where as accurate.

tnakaicode, 2023-09-15:
How about
```
def execute(points):
    # Creation of points for the spine
    # source: https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_pipe.py
    from OCC.Core.TColgp import TColgp_Array1OfPnt
    from OCC.Core.Geom import Geom_BezierCurve, Geom_BSplineCurve, Geom_Circle
    from OCC.Core.GeomFill import GeomFill_Pipe
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeShell
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipeShell
    from OCC.Core.Law import Law_Linear
    from OCC.Extend.Construct import vec_to_dir
    length = len(points)
    array = TColgp_Array1OfPnt(0, length)
    for index in range(length):
        point = points[index]
        print(point)
        array.SetValue(index, gp_Pnt(point[0], point[1], point[2]))

    # Creation of a Bezier Curve as the spine
    bz_curv = Geom_BezierCurve(array)
    bz_curv_edge = BRepBuilderAPI_MakeEdge(bz_curv).Edge()
    bz_curv_wire = BRepBuilderAPI_MakeWire(bz_curv_edge).Wire()

    # Creation of profile to sweep along the spine
    radius = 1.60
    axis = gp_Ax2(array.Value(0), vec_to_dir(gp_Vec(array.Value(0), array.Value(1))))
    circle = gp_Circ(axis, radius)
    circle_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
    circle_wire = BRepBuilderAPI_MakeWire(circle_edge).Wire()
    
    # pipe
    pipe = BRepOffsetAPI_MakePipe(bz_curv_wire, circle_edge).Shape()
    display.DisplayShape(pipe)
    display.DisplayShape(circle_edge)
```

I think bz_curve is pretty crezy, too.

nsmela, 2023-09-15:
thanks, but that gives me a "NotDone" fault when generating the shape. I think I figured it out though:

[image]

My issue was that I didn't know what the points in the bezier curve were doing: I though I was adding points the curve would align to and what I was actually doing was adding control points to generate a curve. So I used a few points to create the curve and used the rest to generate the straight line sections

So even though the fillet idea you posted wasn't exactly what I was looking for, it was along the right track. 
```
def generate(points): 
    # ref: https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_pipe.py
    # ref: https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_edge.py
    from OCC.Core.gp import gp_Pnt, gp_Pln, gp_Dir, gp_Circ, gp_Ax2
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipe, BRepOffsetAPI_ThruSections
    from OCC.Core.Geom import Geom_BezierCurve, Geom_Circle
    from OCC.Core.ChFi2d import ChFi2d_AnaFilletAlgo
    from OCC.Core.GeomFill import GeomFill_Pipe
    from OCC.Core.TColgp import TColgp_Array1OfPnt
    
    # converting points array into gp_Pnt array
    p1 = gp_Pnt(points[0][0], points[0][1], points[0][2])
    p2 = gp_Pnt(points[-2][0], points[-2][1], points[-2][2]) # end of first straight path, start of curve
    p3 = gp_Pnt(points[-1][0], points[-1][1], points[-1][2]) # middle of curve
    p4 = gp_Pnt(points[-1][0], points[-1][1], points[-1][2] - 3.0) # end of curve, start of second straight path
    p5 = gp_Pnt(points[-1][0], points[-1][1], -1.0) #end of second straight path directly downwards
    
    # the path will be broken down into three edges: 
    # first is a straight path from the start of the points to close to the end
    # second is a bezier curve joining the two staight paths
    # third is a straight path down
    
    edge1 = BRepBuilderAPI_MakeEdge(p1, p2).Edge()
    
    # curve joining two straight paths
    array = TColgp_Array1OfPnt(1, 3)
    array.SetValue(1, p2)
    array.SetValue(2, p3)
    array.SetValue(3, p4)
    bz_curve = Geom_BezierCurve(array)
    edge2 = BRepBuilderAPI_MakeEdge(bz_curve).Edge()
    
    # straight path directly downwards
    edge3 = BRepBuilderAPI_MakeEdge(p4, p5).Edge()

    # assembling the path
    wire = BRepBuilderAPI_MakeWire()
    wire.Add(edge1)
    wire.Add(edge2)
    wire.Add(edge3)
    wire.Build()
    
    # circle profile
    direction = gp_Dir(0,0,1)
    vector = gp_Ax2(p3, direction)
    radius = 1.60
    circle = Geom_Circle(vector, radius)
    circle_edge = BRepBuilderAPI_MakeEdge(circle).Edge()
    profile = BRepBuilderAPI_MakeWire(circle_edge).Wire()
    
    pipe = BRepOffsetAPI_MakePipe(wire.Wire(), profile)
    pipe.Build()
    # pipe.MakeSolid()
    return pipe.Shape()

data = [
    [-33.708023525841, -2.496432489532598, 206.2113114751896],
    [-33.325328584549, -4.014806351862397, 201.4628290443381],
    [-32.9426336432576, -5.5331802141921, 196.71434661348653],
    [-32.559938701965905, -7.051554076521899, 191.965864182635],
    [-32.1772437606742, -8.569927938851599, 187.2173817517835],
    [-31.7945488193825, -10.088301801181299, 182.46889932093188],
    [-31.411853878090803, -11.606675663511098, 177.72041689008],
    [-31.0291589367991, -13.125049525840797, 172.971934459229],
    [-30.6464639955074, -14.6434233881706, 168.22345202837698],
    [-30.2637690542157, -16.1617972505003, 163.47496959752598],
    [-29.881074112924, -17.68017111283, 158.726487166674]
]

if __name__ == "__main__":
    from OCC.Core.Graphic3d import Graphic3d_NOM_TRANSPARENT
    from OCC.Display.SimpleGui import init_display    
    display, start_display, add_menu, add_function_to_menu = init_display()
    
    bezier_pipe = generate(data) #generate(data)
    display.DisplayShape(bezier_pipe, update = True)
    
    start_display()
```

Now I just need to figure out how to make the radius consistent throughout and to make it solid.

nsmela, 2023-10-05:
This is what I settled on:

```
def generate_curved_channel(channel: NeedleChannel, cylinder_offset: float, diameter: float = 3.0) -> TopoDS_Shape:
    '''
    Generates a TopoDS_Shape from the Needle Channel's points
    Cylinder Offset is for height offset
    diameter is the channel's diameter
    '''
    # offset points using z axis and cylinder's offset
    # and convert into a gp_Pnt
    points = []
    for point in channel.points:
        points.append(gp_Pnt(point[0], point[1], point[2] - cylinder_offset))
    
    radius = diameter /2 
    
    # generate starting point on top (cone)
    p1 = points[0]
    p2 = points[1]
    vector = np.array([channel.points[1][0], channel.points[1][1], channel.points[1][2]]) \
        - np.array([channel.points[0][0], channel.points[0][1], channel.points[0][2]])
    length = np.linalg.norm(vector)
    direction = helper.get_direction(p1, p2)
    axis = gp_Ax2(p1, direction)
    pipe = BRepPrimAPI_MakeCone(axis, 0.0, radius, length).Shape()
    face = helper.get_lowest_face(pipe)
    
    # for each (after the first), create a sphere and cylinder to next point to join
    for i in range(1, len(points) - 1):
        p1 = points[i]
        p2 = points[i + 1]
        
        edge = BRepBuilderAPI_MakeEdge(p1, p2).Edge()
        makeWire = BRepBuilderAPI_MakeWire(edge)
        makeWire.Build()
        wire = makeWire.Wire()
        cylinder = BRepOffsetAPI_MakePipe(wire, face).Shape()
        pipe = BRepAlgoAPI_Fuse(pipe, cylinder).Shape()
        face = helper.get_lowest_face(cylinder)

    # add a curved pipe downwards using offset length and direction of last two points
    vector = helper.get_vector(points[-2], points[-1], length)
    p1 = points[-1]
    p2 = gp_Pnt(p1.X() + vector.X(), p1.Y() + vector.Y(), p1.Z() + vector.Z())
    p3 = gp_Pnt(p2.X(), p2.Y(), p2.Z() - length)
    
    # curve joining two straight paths
    array = TColgp_Array1OfPnt(1, 3)
    array.SetValue(1, p1)
    array.SetValue(2, p2)
    array.SetValue(3, p3)
    bz_curve = Geom_BezierCurve(array)
    bend_edge = BRepBuilderAPI_MakeEdge(bz_curve).Edge()
    
    # assembling the path
    wire = BRepBuilderAPI_MakeWire(bend_edge).Wire()
    
    # shape using last face
    pipe_bend = BRepOffsetAPI_MakePipe(wire, face).Shape()
    pipe = BRepAlgoAPI_Fuse(pipe, pipe_bend).Shape()
    
    # add a cylinder from pipe to past bottom of cylinder 
    base_point = gp_Pnt(p3.X(), p3.Y(), -0.01)
    face = helper.get_lowest_face(pipe_bend)
    edge = BRepBuilderAPI_MakeEdge(p3, base_point).Edge()
    makeWire = BRepBuilderAPI_MakeWire(edge)
    makeWire.Build()
    wire = makeWire.Wire()
    cylinder = BRepOffsetAPI_MakePipe(wire, face).Shape()
    pipe = BRepAlgoAPI_Fuse(pipe, cylinder).Shape()

    return pipe
```

Here are the helper functions
```
def face_is_plane(face: TopoDS_Face) -> bool:
    """
    Returns True if the TopoDS_Face is a plane, False otherwise
    """
    surf = BRepAdaptor_Surface(face, True)
    surf_type = surf.GetType()
    return surf_type == GeomAbs_Plane

def geom_plane_from_face(face: TopoDS_Face) -> gp_Pln:
    """
    Returns the geometric plane entity from a planar surface
    """
    return BRepAdaptor_Surface(face, True).Plane()

def get_faces(shape: TopoDS_Shape) -> list[TopoDS_Face]:
    '''
    returns a list of planar faces for the shape from highest to lowest
    '''
    def sortByZ(elem):
        return elem[1]
    
    explorer = TopExp_Explorer(shape, TopAbs_FACE)
    faces = []
    while explorer.More():
        face = topods.Face(explorer.Current())
        if face_is_plane(face):
            a_plane = geom_plane_from_face(face)
            faces.append([face, a_plane.Location().Z()]) # face with it's z height
        explorer.Next()
    faces.sort(key=sortByZ)
    return faces

def get_highest_face(shape: TopoDS_Shape) -> TopoDS_Face:
    faces = get_faces(shape)
    return faces[-1][0]

def get_lowest_face(shape: TopoDS_Shape) -> TopoDS_Face:
    faces = get_faces(shape)
    return faces[0][0]

def get_vector(p1: gp_Pnt, p2: gp_Pnt, length: float = 1.0) -> gp_Vec:
    vector = gp_Vec(
        p2.X() - p1.X(),
        p2.Y() - p1.Y(),
        p2.Z() - p1.Z())
    return vector.Normalized() * length

def get_direction(p1: gp_Pnt, p2: gp_Pnt) -> gp_Dir:
    vector = get_vector(p1, p2)
    return gp_Dir(vector.X(), vector.Y(), vector.Z())
````
it produced these purple channels:
[image]

So the overall process:

- create a cone and save the lowest face as a profile for the next point
- iterate through the points, generating a cylinder and saving the lowest face for the next portion of the pipe
- at the end, make bend using three points: the last point, a projected point, and a point directly below that projected point)
- finally, extend the last face down past the base
- fused the shapes along the way 

Thank you for the advice!

closed 2023-10-05 by nsmela