tpaviot/pythonocc-core #1249: How to create Standard_Stream in pyocc

ICO-Fantasy, 2023-09-25:
Python has its own way of handling strings, and I realized that in pyocc the gp_Trsf object lacks the DumpJson method, but has the InitFromJson method, which receives a `Standard_Stream const &`, as shown in the error report
`TypeError: in method 'gp_Trsf_InitFromJson', argument 2 of type 'Standard_SStream const &'`

I don't know how to create a Standard_SStream to populate gp_Trsf with data. 
 I guess it should be filled with a Json string like this:
`"Location": [10.0, 20.0, 30.0]"Matrix": [-0.598, 0.801, 0.0, -0.801, -0.598, 0.0, 0.0, 0.0, 1.0]"shape": 7 "scale": 1`

I'm trying to implement a DumpJson myself to get the desired json string, and I need to utilize InitFromJson for testing.
Does anyone know how to create a Standard_Stream object? thanks a lot!

tpaviot, 2023-10-01:
The DumpJson method is wrapped as `DumpJsonToString`. I had a deep look at the occt json serializer, but I can't understand how it works, and I can't even write a simple c++ example. Strings are not correctly json formatted. IMO if you need to serialize objects, you can implement a function by yourself that uses the json pytho nmodule, it will be safer.

tpaviot, 2023-10-01:
you're right Standard_SStream is not wrapped, it is actually used only for `ÌnitFromJson' method

tpaviot, 2023-10-02:
I tried with the following c++ example, it seemds the `InitFromJson` fails at modifying parameters, unless I'm doing something wrong:
```c++
#include <iostream>
#include <gp_Trsf.hxx>

/* to compile :
$ g++ -I /opt/occt772/include/opencascade/ -L /opt/occt772/lib essai_json.cpp -o essai_gp -lTKernel -lTKMath
*/
int main () {
    gp_Trsf p;
    std::stringstream s;
    p.SetScaleFactor(2.5);
    p.DumpJson(s);
    std::cout << s.str() << std::endl;
	
    gp_Trsf other_trsf;
    int pos;
    other_trsf.InitFromJson(s, pos);
    std::cout << "Scale factor: " << other_trsf.ScaleFactor() << std::endl;
    return 0;
}
```
ouput:
```bash
"Location": [0, 0, 0], "Matrix": [1, 0, 0, 0, 1, 0, 0, 0, 1], "shape": 6, "scale": 2.5
Scale factor: 1
```

tpaviot, 2023-10-06:
Fixed by PR #1258, which adds pickling through json serialization

closed 2023-10-06 by tpaviot