tpaviot/pythonocc-core #1253: Segmentation fault and other errors in BRepAlgoAPI_Cut

stepi9, 2023-09-29:
I'm using pythonocc-core compiled from source (version 7.7.2) to perform cuts of cylinders (with radius of order 10^1 and length of order 10^2), organised in a cubic lattice, from a complex 3d step geometry of a bone sample (dimension order: 10^3), reconstructed from stl using Fusion 360. I'm using BRepAlgoAPI_Cut but I get a lot of errors and/or warnings. If I'm lucky, the program finishes, but some expected cuts are missing and/or some artifacts are present (During the execution I may get several warnings from brep cut, such as: too small edge, not splittable edge, acquired self intersection and others). If I'm less lucky, at a certain point, the geometry is sort of corrupted with only a few particles remaining. In the worst case I get an Exception from the cut function or even a Segmentation Fault. I've also tried with fuzzy value and shapeupgrade but with no to limited benefits. What would you suggest to solve this issue?

tpaviot, 2023-10-01:
thank you for your feedback. Hard to answer without a bit more information, ensure the shape from stl conversion is consistent and produces a clean TopoDS_Shape.

stepi9, 2023-10-01:
> ensure the shape from stl conversion is consistent and produces a clean TopoDS_Shape.

I've tried to import the STEP file in Autodesk Inventor, Solidworks, Gmsh and FreeCAD and neither of them gave me any error