tpaviot/pythonocc-core #1254: Can qtViewer3d be run in a child process?

ovo-Tim, 2023-10-02:
When qtViewer3d and QT are in the same process, the rendering of OCC makes the qt interface laggy. Is it possible to separate them into two processes? The rendering of qtViewer3d onto the Qt widget mainly relies on winID, so there is no complex information transfer between them. I think it is possible.

I am trying to write a library that allows code to be migrated to a child process without modification, but the data transfer overhead is too large and I encountered the problem `TypeError: cannot pickle 'SwigPyObject' object`. I am thinking of a better solution.

tpaviot, 2023-10-03:
I don't know how to answer your question, I'm not a qt expert. What kind of object do you need to pickle?

ovo-Tim, 2023-10-03:
> I don't know how to answer your question, I'm not a qt expert. What kind of object do you need to pickle?

Thanks for your reply. I have written a [python library](https://github.com/ovo-Tim/multiprocessing-mq), but I can't pickle any OCC object(They actually is `SwigPyObject`).

tpaviot, 2023-10-04:
pickling is implemented only for `TopoDS_Shape`, see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_shape_pickling.py 

If you need to pickle other objects, I see two solutions :
* use the methods `DumpJson`  and `InitFromJson` if they are available. Only a few classes provide these methods, dumped string is not clean json, and furthermore I can't get the InitFomJson work
* implement your own json serializer/deserializer but unfortunately you cannot do that for all occt classes

ovo-Tim, 2023-10-04:
Thanks, it may solve some problems. However, I think just create OCC object in child process is the best solution, because you don't need to copy objects from this process to other processes. But it also means you have to rewrite a lot of things.

tpaviot, 2023-10-06:
Fixed by PR #1258

closed 2023-10-06 by tpaviot

ovo-Tim, 2023-10-15:
It shouldn't be close. I'm still trying to do that, but it's really hard.

ovo-Tim, 2023-10-15:
I have done some research. Sadly, I have to say that's almost impossible. But you still can use multiprocessing in some specific scenes like loading file.