tpaviot/pythonocc-core #1259: Quick Shape Intersecting Test

nsmela, 2023-10-08:
Is there a simple or effective way to detect if a shape or solid is overlapping one or more other shapes from other solids? Something with a boolean result? Im assuming such a check is done as part of the boolean operations, but I couldn't find the right function.

tpaviot, 2023-10-09:
I guess the `BRepAlgoAPI_Common` is not enough for you.

I would mesh the shape using the `ShapeTesselator` class, and use a dedicated library for the collision check, for instance https://github.com/humanoid-path-planner/hpp-fcl

nsmela, 2023-10-10:
Would you recommend trying to use BRepAlgoAPI_Common by combining the two and if there's a generated shape, the two are intersecting? Or is that a really poor approach?

tpaviot, 2023-10-10:
This has to be benchmarked. IMO using meshes and a collision library will be better.

closed 2023-10-10 by nsmela

Tanneguydv, 2023-10-11:
> Would you recommend trying to use BRepAlgoAPI_Common by combining the two and if there's a generated shape, the two are intersecting? Or is that a really poor approach?

If time is not an issue for you I suggest to use BRepAlgoAPI_Section and proceed like this:
```collision_shapes = BRepAlgoAPI_Section(part1, part2)
        if collision_shapes.NbChildren() > 0:
            print("shapes are intersecting")
        else:
            print("no intersection found")
```

nsmela, 2023-10-11:
> If time is not an issue for you I suggest to use BRepAlgoAPI_Section and proceed like this:
>
> ```
> collision_shapes = BRepAlgoAPI_Section(part1, part2)
> if collision_shapes.NbChildren() > 0: [...]

Tried this, "BRepAlgoAPI_Section object has no attribute NbChildren"

nsmela, 2023-10-11:
Ok, this works:
```
from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Section
from OCC.Core.TopoDS import TopoDS_Shape

def are_colliding(shape1:TopoDS_Shape, shape2: TopoDS_Shape) -> bool:
    collision_shapes = BRepAlgoAPI_Section(shape1, shape2)
    return collision_shapes.Shape().NbChildren() > 0
```

That should make it easy to benchmark later when we try other methods later. Thank you both for your help!