tpaviot/pythonocc-core #1267: conda-forge packages for 7.7.2

tpaviot, 2023-10-25:
pythonocc-core-7.7.2 is released. todo : build and publish 7.7.2 binaries to anaconda

ping @looooo

lanhao945, 2023-10-25:
I just have a question. (Sorry,English is not my native language,I asked the question respectfully and suspiciously. Not trying to convey any unkindness)

In the readme file, it says, we can install the package by:

```shell
conda install -c conda-forge pythonocc-core=7.7.2
```

But in fact，the anaconda's conda-forge,just  publish the 7.7.0.

Could anyone change the document in the readme file?or , some other reason, couldn‘t change it ?

(About one year ago, have the same status quo, the version in readme file bigger than the version in anaconda）.

rainman110, 2023-10-25:
The reason is, that pythonocc 7.7.2 is just released and with it it's installation instructions. To be also available on conda-forge @looooo needs to update the conda package also, which is a separate step, requiring also another merge request on conda-forge. Please wait for a few days, then it will be available.

lanhao945, 2023-10-25:
Thank you very much~~

tpaviot, 2023-10-25:
@lanhao945 I tagged the 7.7.2 release this morning, it's just a question of days to get the conda-forge package available. In case of emergency, you can still compile it by yourself

rainman110, 2023-10-25:
There are already 3 PRs for pythonocc at conda-forge ;) : https://github.com/conda-forge/pythonocc-core-feedstock/pulls

tpaviot, 2023-10-28:
conda packages were uploaded

closed 2023-10-28 by tpaviot