tpaviot/pythonocc-core #1268: SkipShared = True calculates too many edges for a shared edge

rozzzi, 2023-10-25:
Hi,

When using the LinearProperties function from BRepGProp to calculate the length of the edges of a unit cube, the total length of the edges should be 24 if Skipshared=False and 12 if SkipShared=True, because every edge is shared by two faces. SkipShared=True should take edges that are shared by two or more faces only once according to the OCC documentation (https://dev.opencascade.org/doc/refman/html/class_b_rep_g_prop.html). 
When using this function to calculate the edges of a TopoDS_Compound shape that consists of 2 unit cubes that share 1 face exactly (resulting in a 1x1x2 shape), one would assume that with SkipShared = False this would result in a total edge length of 24 + 24 = 48, as the middle edges are shared by both cubes but will still be counted for every face it is an edge of.
With SkipShared = True, one would expect a total edge length of 8 + 4 + 8 = 20, as the middle 4 edges are shared by 4 faces each, but should only be counted once. However, the outcome of the function is 24. Presumably this is because the edges shared by faces within unit cubes are only counted once, but when the edges overlap they are counted individually, even though they are in fact 1 edge, and share two or more faces but should only be counted once. Alternatively, one could state that these are not edges anymore, because the shape has become 1 big shape, and the shared face is not a face anymore (so the total edge length would be 4+8+4 = 16).
Is this how the code is supposed to work? If so, is there a different function that would essentially do what I am trying to calculate?

Thank you.

tpaviot, 2023-10-25:
Related code is here: https://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=src/BRepGProp/BRepGProp.cxx;h=43c9d08129c913e1eb83963c53d889dc60582146;hb=HEAD#l83

Basically, a hash table allows getting unique TopoDS_Edges. There's not special topological algorithm, only the TopoDS_Edge hash is taken into account to remove duplicates.

For two cubes that do not share any face, is the result correct? Feel free to share a code snippet to support the discussion

rozzzi, 2023-10-25:
The issue might not be SkipShared itself, but rather the question as what would produce the right results. Here is the file that consist of 2 cubes with a shared face:
[array3_2mm3_area_edge.txt](https://github.com/tpaviot/pythonocc-core/files/13164525/array3_2mm3_area_edge.txt)

```
from OCC.Core.GProp import GProp_GProps

def occ_core_edge_length_skipshared(shape): #, Properties) -> float:
    """Calculates the edge length of input shape

    Parameters
    ----------
    shape : TopoDS_Shape / TopoDS_Solid

    Returns
    -------
    Length : scalar
        Calculated edge length
    """
    Properties = GProp_GProps()
    brepgprop_LinearProperties(shape, Properties, True)

    Length_skipshared = Properties.Mass()

    return Length_skipshared
    
def occ_core_surface_area_skipshared(shape): #, Properties) -> float:
    """Calculates the surface area of input shape
    ! ! ! SkipShared = False: Calculates also the overlapping surfaces

    Parameters
    ----------
    shape : TopoDS_Shape / TopoDS_Solid

    Returns
    -------
    Area : scalar
        Calculated surface area
    """
    Properties = GProp_GProps()
    brepgprop_SurfaceProperties(shape, Properties, True)

    Area_skipshared = Properties.Mass()

    return Area_skipshared
```
    
With this code, the returned edge length is 24 and the returned surface area is 12. What I am looking for is a way to return 20 and 10 respectively.

tnakaicode, 2023-11-27:
The results depend on how you connect the two boxes.
I tried of the three methods (MakeCompound/Sewing/Fuse), and gave reasonable results only by Fuse.
I did not verify if part of the Edge, part of the Face, or part of the Volume was shared, or if more than 3 Boxes were connected.

```python
from OCC.Core.GProp import GProp_GProps
from OCC.Core.BRepGProp import brepgprop
from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file
from OCC.Core.gp import gp_Pnt, gp_Dir, gp_Ax3, gp_Ax2
from OCC.Core.TopoDS import TopoDS_Shell, TopoDS_Compound
from OCC.Core.BOPAlgo import BOPAlgo_Builder
from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing
from OCC.Core.TopTools import TopTools_ListOfShape
from OCC.Core.LocOpe import LocOpe_FindEdges, LocOpe_FindEdgesInFace
from OCCUtils.Construct import make_box

def property_of_box(box=make_box(1, 1, 1)):
    skip_shared = True
    use_triangle = False
    only_closed = True

    print(box)
    print("Skip Shared: ", skip_shared)

    boxs_prop_line = GProp_GProps()
    brepgprop.LinearProperties(box,
                               boxs_prop_line,
                               skip_shared, use_triangle)
    print("Mass of Lines", boxs_prop_line.Mass())

    boxs_prop_surf = GProp_GProps()
    brepgprop.SurfaceProperties(box,
                                boxs_prop_surf,
                                skip_shared, use_triangle)
    print("Mass of Surfaces", boxs_prop_surf.Mass())

    boxs_prop_volm = GProp_GProps()
    brepgprop.VolumeProperties(box,
                               boxs_prop_volm,
                               only_closed, skip_shared, use_triangle)
    print("Mass of Volume", boxs_prop_volm.Mass())

if __name__ == "__main__":
    display, start_display, add_menu, add_function_to_menu = init_display()

    box1 = make_box(gp_Pnt(0, 0, 0), gp_Pnt(1, 1, 1))
    box2 = make_box(gp_Pnt(0, 0, 0), gp_Pnt(-1, -1, -1))
    box3 = make_box(gp_Pnt(1, 0, 0), gp_Pnt(2, 1, 1))
    box4 = make_box(gp_Pnt(0, 0, 0), gp_Pnt(1, -1, -1))
    box5 = make_box(gp_Pnt(0.5, 0, 0), gp_Pnt(1.5, -1, -1))

    boxs = [box1, box3]

    # Make Compound by two boxes
    boxs_comp = TopoDS_Compound()
    bild = BRep_Builder()
    bild.MakeCompound(boxs_comp)
    for shp in boxs:
        bild.Add(boxs_comp, shp)
    property_of_box(boxs_comp)  # Line: NG, Surface: NG

    # Make Shell by sewing two boxes
    sew = BRepBuilderAPI_Sewing()
    for shp in boxs:
        sew.Add(shp)
    sew.Perform()
    boxs_sewed = sew.SewedShape()
    property_of_box(boxs_sewed)  # Line: NG, Surface: NG

    # Make Shape by fuse two boxes
    fuse = BOPAlgo_Builder()
    for shp in boxs:
        fuse.AddArgument(shp)
    fuse.Perform()
    boxs_fuse = fuse.Shape()
    property_of_box(boxs_fuse)  # Line: OK, Surface: OK, Volume: OK

    # Find Edge that the two boxes share
    find_edge = LocOpe_FindEdges(boxs[0], boxs[1])
    find_edge.InitIterator()
    while find_edge.More():
        display.DisplayShape(find_edge.EdgeTo(), color="BLUE1")
        display.DisplayShape(find_edge.EdgeFrom(), color="RED")
        find_edge.Next()

    def display_box(box=make_box(1, 1, 1), name="box"):
        display.DisplayShape(box, transparency=0.9)
        prop = GProp_GProps()
        brepgprop.SurfaceProperties(box, prop, True, False)
        pnt = prop.CentreOfMass()
        display.DisplayMessage(pnt, name)

    display_box(box=box1, name="box1")
    display_box(box=box2, name="box2")
    display_box(box=box3, name="box3")
    display_box(box=box4, name="box4")
    display_box(box=box5, name="box5")
    display.FitAll()
    start_display()

```