tpaviot/pythonocc-core #1275: Wire extraction after MakeFillet2d not following the Face

Squirrel8475, 2023-11-07:
Hello,

I have a model with a Wire and need to add a fillet to the Wire. To do this, a Face is created from the Wire, the Face is filleted and the outer wire of this filleted face is extracted. When the face has some 'concavity' (better said the center of the arc is outside the face),  as illustrated below, the resulting Wire does not follow the Face contour. It looks like the wire is flipped locally. Is it the expected behavior? Any idea to fix that?

- Pink : wires before fillet
- Grey : face after fillet
- Blue: wire extraction after fillet (not following the face)

 
[image]

Code to produce the picture above:

````python
from OCC.Core.BRepBuilderAPI import (BRepBuilderAPI_MakeVertex,\
BRepBuilderAPI_MakeEdge,BRepBuilderAPI_MakeWire,BRepBuilderAPI_MakeFace)
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet2d
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopAbs import TopAbs_WIRE
from OCC.Core.BRepTools  import breptools
from OCC.Core.gp import gp_Pnt

coords=[
    (0, 50, 0),
    (50,50, 0),
    (50,25, 0),
    (80,25, 0),
    (80,0, 0),
    (0,0, 0)

]

# Create vertices
vertices=list()
for coord in coords:
    vertices.append( BRepBuilderAPI_MakeVertex(gp_Pnt(*coord)).Shape() )

# Join vertices to create a closed loop of edges
nv=len(vertices)
edges=list()
for i in range(0,nv,1):
    if i<nv-1:
        edge=BRepBuilderAPI_MakeEdge(vertices[i],vertices[i+1]).Shape()
    elif i==nv-1:
        edge=BRepBuilderAPI_MakeEdge(vertices[i],vertices[0]).Shape()
    else:
        raise ValueError('Out of bounds, i=',i)
    edges.append(edge)

# Create the wire and the corresponding face
a_maker = BRepBuilderAPI_MakeWire()
for edge in edges:
    a_maker.Add(edge)
a_maker.Build()
original_topods_wire=a_maker.Shape()

OnlyPlane = True
a_maker = BRepBuilderAPI_MakeFace(original_topods_wire,OnlyPlane)
a_maker.Build()
a_topods_face = a_maker.Shape()

# Make Fillet
a_maker=BRepFilletAPI_MakeFillet2d(a_topods_face)
a_maker.AddFillet(vertices[2],10)
a_maker.Build()
a_topods_face=a_maker.Shape()

# Get the wire out of the face
option=1
if option==0:
    an_explorer=TopExp_Explorer(a_topods_face,TopAbs_WIRE)
    i=0
    while an_explorer.More(): 
        print(i)
        a_topods_wire=an_explorer.Value()
        an_explorer.Next()
        i=i+1
elif option==1:
    a_topods_wire=breptools.OuterWire(a_topods_face)

my_renderer = JupyterRenderer() 
my_renderer.DisplayShape(original_topods_wire, render_edges=True,update=False,edge_color='magenta')
my_renderer.DisplayShape(a_topods_wire, render_edges=True,update=False)
my_renderer.DisplayShape(a_topods_face, render_edges=False,update=True)

````

Tanneguydv, 2023-11-09:
Hello,
I guess it's only a visualization problem from JupyterRenderer, here the output of your code using SimpleGui :

[image]

The code : 
``` 
from OCC.Core.BRepBuilderAPI import (BRepBuilderAPI_MakeVertex,\
BRepBuilderAPI_MakeEdge,BRepBuilderAPI_MakeWire,BRepBuilderAPI_MakeFace)
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet2d
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopAbs import TopAbs_WIRE
from OCC.Core.BRepTools  import breptools
from OCC.Core.gp import gp_Pnt

from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

coords=[
    (0, 50, 0),
    (50,50, 0),
    (50,25, 0),
    (80,25, 0),
    (80,0, 0),
    (0,0, 0)

]

# Create vertices
vertices=list()
for coord in coords:
    vertices.append( BRepBuilderAPI_MakeVertex(gp_Pnt(*coord)).Shape() )

# Join vertices to create a closed loop of edges
nv=len(vertices)
edges=list()
for i in range(0,nv,1):
    if i<nv-1:
        edge=BRepBuilderAPI_MakeEdge(vertices[i],vertices[i+1]).Shape()
    elif i==nv-1:
        edge=BRepBuilderAPI_MakeEdge(vertices[i],vertices[0]).Shape()
    else:
        raise ValueError('Out of bounds, i=',i)
    edges.append(edge)

# Create the wire and the corresponding face
a_maker = BRepBuilderAPI_MakeWire()
for edge in edges:
    a_maker.Add(edge)
a_maker.Build()
original_topods_wire=a_maker.Shape()

OnlyPlane = True
a_maker = BRepBuilderAPI_MakeFace(original_topods_wire,OnlyPlane)
a_maker.Build()
a_topods_face = a_maker.Shape()

# Make Fillet
a_maker=BRepFilletAPI_MakeFillet2d(a_topods_face)
a_maker.AddFillet(vertices[2],10)
a_maker.Build()
a_topods_face=a_maker.Shape()

# Get the wire out of the face
option=1
if option==0:
    an_explorer=TopExp_Explorer(a_topods_face,TopAbs_WIRE)
    i=0
    while an_explorer.More(): 
        print(i)
        a_topods_wire=an_explorer.Value()
        an_explorer.Next()
        i=i+1
elif option==1:
    a_topods_wire=breptools.OuterWire(a_topods_face)

display.DisplayShape(original_topods_wire, color=23) #color = blue
display.DisplayShape(a_topods_wire) # color = red
display.DisplayShape(a_topods_face, transparency=0.8)

display.FitAll()
start_display()

```

tpaviot, 2023-11-10:
I confirm this is an issue with the jupyter renderer only

tpaviot, 2023-11-10:
this certainly has to do with the circular edge orientation

tpaviot, 2023-11-10:
Fixed.

[image]

closed 2023-11-10 by tpaviot

Squirrel8475, 2023-11-10:
Many thanks! @tpaviot @Tanneguydv

commit 137c61de (2023-11-10): Fix wrong edge orientation in discretize_edge, fix #1275