tpaviot/pythonocc-core #1278: Setting label name with umlaut result in STEP with encoding issues, trying extended string fails with crash

manuel-koch, 2023-11-16:
With pythonocc-core=7.7.2 , trying to export XCAF to STEP, using label names.
There is an encoding issue with names that contain e.g. umlaut characters.
Trying to workaround the issue by using fails too.

How can I fix this encoding issue, how to use pythonocc-core properly ?

Minimal working sample code

```python
from OCC.Core import TopoDS, TopAbs
from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.BinXCAFDrivers import binxcafdrivers
from OCC.Core.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCC.Core.STEPCAFControl import STEPCAFControl_Writer
from OCC.Core.TCollection import TCollection_ExtendedString
from OCC.Core.TDF import TDF_Label
from OCC.Core.TDataStd import TDataStd_Name
from OCC.Core.TDocStd import TDocStd_Application, TDocStd_Document
from OCC.Core.TopoDS import TopoDS_Shape, TopoDS_Compound
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_ColorGen
from OCC.Extend.DataExchange import read_step_file
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Core.gp import gp_Trsf, gp_Vec, gp_Quaternion
from OCC.Core.STEPControl import (
    STEPControl_AsIs,
)
from OCC.Core.Interface import Interface_Static
from OCC.Core.IFSelect import IFSelect_RetDone

# Create XDE document
app = TDocStd_Application()
binxcafdrivers.DefineFormat(app)
doc = TDocStd_Document(f"example")
app.NewDocument("BinXCAF", doc)

# Tools
shape_tool = XCAFDoc_DocumentTool.ShapeTool(doc.Main())  # XCAFDoc_ShapeTool
color_tool = XCAFDoc_DocumentTool.ColorTool(doc.Main())  # XCAFDoc_ColorTool

compound = TopoDS_Compound()
brep_builder = BRep_Builder()
brep_builder.MakeCompound(compound)

shape = BRepPrimAPI_MakeBox(10, 10, 10).Shape()
label = shape_tool.AddShape(shape, False)

# Set custom name on label ( i.e. the shape it refers to )
# this has encoding issues
#   TDataStd_Name.Set(label, "Some text with umlauts äöü")
# BUT the following crashes with
# TypeError: Wrong number or type of arguments for overloaded function 'TDataStd_Name_Set'.
#     Possible C/C++ prototypes are:
#         TDataStd_Name::Set(TDF_Label const &,TCollection_ExtendedString)
#         TDataStd_Name::Set(TDF_Label const &,Standard_GUID const &,TCollection_ExtendedString)
#         TDataStd_Name::Set(TCollection_ExtendedString)
extended_string = TCollection_ExtendedString()
TDataStd_Name.Set(label, extended_string)

# add moved instance instance of a prototype to our compound shape
brep_builder.Add(compound, shape)

compund_label = shape_tool.AddShape(compound, True)

# Set custom name on label ( i.e. the shape it refers to )
TDataStd_Name.Set(compund_label, "compound")

# Initialize the STEP exporter
step_writer = STEPCAFControl_Writer()

# To make sub-shape names work, we have to turn on the following static
# variable of OpenCascade.
Interface_Static.SetIVal("write.stepcaf.subshapes.name", 1)

Interface_Static.SetCVal("write.step.schema", "AP214")
Interface_Static.SetCVal("write.step.product.name", "my product")

# transfer compound shape and write STEP file
step_writer.Transfer(doc, STEPControl_AsIs)
output_step_path = ASSETS_DIR / "compound_with_umlaut_label.step"
status = step_writer.Write(output_step_path.as_posix())

if status != IFSelect_RetDone:
    raise AssertionError("write failed")
```

The generated STEP looks like the following in CAD Assistent:
[image]

The generated STEP ( with the encoding issue ):
[compound_with_umlaut_label.step.txt](https://github.com/tpaviot/pythonocc-core/files/13380429/compound_with_umlaut_label.step.txt)

tpaviot, 2023-11-17:
thank you for the feedback, fixed in current master

closed 2023-11-17 by tpaviot

manuel-koch, 2023-11-17:
Will there be a new release available for conda or regular python environments ? I tried to build pythonocc-core for my regular python environment but failed due to build problems of dependencies.