tpaviot/pythonocc-core #1281: MakeFillet between 2 faces is Failure

tnakaicode, 2023-11-22:
When I tried to create a Fillet between two faces with shared edges, it failed at the fillet.Build() stage.
In my code below, Fillet is created by taking 2 faces that share an edge from a box created by MakeBox.
It succeeds in box1, but fails in box2 when the faces are repositioned (rotated 0 degrees) and there are clearly shared edges.

```python
import sys
import numpy as np
from math import cos, pi

from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet
from OCC.Core.TopoDS import TopoDS_Shell
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Display.SimpleGui import init_display
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir, gp_Pnt2d, gp_Ax1, gp_Trsf
from OCC.Core.LocOpe import LocOpe_FindEdges
from OCC.Extend.TopologyUtils import TopologyExplorer
from OCCUtils.Construct import make_box

display, start_display, add_menu, add_function_to_menu = init_display()

box1 = make_box(gp_Pnt(0, 0, 0), 20, 20, 20)
box1_faces = list(TopologyExplorer(box1).faces())

# Make Shell by only two faces
box1_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box1_shell)
for shp in [box1_faces[0], box1_faces[2]]:
    bild.Add(box1_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box1_faces[0], box1_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box1_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build()
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")
display.DisplayShape(box1_shell, transparency=0.7)

box2 = make_box(gp_Pnt(40, 0, 0), 20, 20, 20)
box2_faces = list(TopologyExplorer(box2).faces())

# Rotate 0 degree
box2_trsf = gp_Trsf()
box2_trsf.SetRotation(gp_Ax1(gp_Pnt(40, 0, 0), gp_Dir(0, 0, 1)), np.deg2rad(0))
box2_faces[0].Move(TopLoc_Location(box2_trsf))

# Make Shell by only two faces
box2_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box2_shell)
for shp in [box2_faces[0], box2_faces[2]]:
    bild.Add(box2_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box2_faces[0], box2_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box2_shell, 0)
fillet.Add(5, fillet_edge)
# fillet.Build() 
# RuntimeError: Standard_Failure There are no suitable edges for chamfer or fillet raised from method Build of class BRepBuilderAPI_MakeShape
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")

display.DisplayShape(box2_shell, transparency=0.7)

display.FitAll()
start_display()
```

Tanneguydv, 2023-11-22:
My thought is that the` BRepFilletAPI_MakeFillet` method can add a fillet to one of the edges owned by the _shape_ passed in arg. Differents solutions:
(I have added a 45 degrees rotation for better understanding). 

1. You could use the` .Move` method for _box2_ instead of _box2_faces[0]_(?), and then list the faces of your moved _box2_. The `.Build()` now works

```
import sys
import numpy as np
from math import cos, pi

from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet
from OCC.Core.TopoDS import TopoDS_Shell
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Display.SimpleGui import init_display
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir, gp_Pnt2d, gp_Ax1, gp_Trsf
from OCC.Core.LocOpe import LocOpe_FindEdges
from OCC.Extend.TopologyUtils import TopologyExplorer
from OCCUtils.Construct import make_box

display, start_display, add_menu, add_function_to_menu = init_display()

box1 = make_box(gp_Pnt(0, 0, 0), 20, 20, 20)
box1_faces = list(TopologyExplorer(box1).faces())

# Make Shell by only two faces
box1_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box1_shell)
for shp in [box1_faces[0], box1_faces[2]]:
    bild.Add(box1_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box1_faces[0], box1_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box1_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build()
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")
display.DisplayShape(box1_shell, transparency=0.7)

box2 = make_box(gp_Pnt(40, 0, 0), 20, 20, 20)

# Rotate 45 degree
box2_trsf = gp_Trsf()
box2_trsf.SetRotation(gp_Ax1(gp_Pnt(40, 0, 0), gp_Dir(0, 0, 1)), np.deg2rad(45))

box2.Move(TopLoc_Location(box2_trsf))
box2_faces = list(TopologyExplorer(box2).faces())

# Make Shell by only two faces
box2_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box2_shell)
for shp in [box2_faces[0], box2_faces[2]]:
    bild.Add(box2_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box2_faces[0], box2_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box2_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build() 
# RuntimeError: Standard_Failure There are no suitable edges for chamfer or fillet raised from method Build of class BRepBuilderAPI_MakeShape
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")

display.DisplayShape(box2_shell, transparency=0.7)

display.FitAll()
start_display()
```
2. You don't want to move your whole _box2_, so you move the _box2_shell_ that you created, then you iterate through your shell to find the common edge : 

```
import sys
import numpy as np
from math import cos, pi

from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet
from OCC.Core.TopoDS import TopoDS_Shell
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Display.SimpleGui import init_display
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir, gp_Pnt2d, gp_Ax1, gp_Trsf
from OCC.Core.LocOpe import LocOpe_FindEdges
from OCC.Extend.TopologyUtils import TopologyExplorer
from OCCUtils.Construct import make_box

display, start_display, add_menu, add_function_to_menu = init_display()

box1 = make_box(gp_Pnt(0, 0, 0), 20, 20, 20)
box1_faces = list(TopologyExplorer(box1).faces())

# Make Shell by only two faces
box1_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box1_shell)
for shp in [box1_faces[0], box1_faces[2]]:
    bild.Add(box1_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box1_faces[0], box1_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box1_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build()
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")
display.DisplayShape(box1_shell, transparency=0.7)

box2_ = make_box(gp_Pnt(40, 0, 0), 20, 20, 20)

# Rotate 45 degree
box2_trsf = gp_Trsf()
box2_trsf.SetRotation(gp_Ax1(gp_Pnt(40, 0, 0), gp_Dir(0, 0, 1)), np.deg2rad(45))
box2_faces = list(TopologyExplorer(box2_).faces())

# Make Shell by only two faces
box2_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box2_shell)
for shp in [box2_faces[0], box2_faces[2]]:
    bild.Add(box2_shell, shp)

box2_shell.Move(TopLoc_Location(box2_trsf))
box2_shell_faces = list(TopologyExplorer(box2_shell).faces())

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box2_shell_faces[0], box2_shell_faces[1])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box2_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build() 
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")

display.DisplayShape(box2_shell, transparency=0.7)

display.FitAll()
start_display()
```

tnakaicode, 2023-11-23:
I am considering the possibility of creating a Fillet with any 2 sides that share an edge.
Your solution is the same situation as Box1 because you are rotating the entire Box to create the 2 sides.
Is there any way to create a Fillet with 2 sides if, for example, I change the rotation of the first code to 30 degrees?

[image]

Tanneguydv, 2023-11-23:
You can't make a fillet out of two single faces, you could extend them and join them with other methods I guess.

`BRepFilletAPI_MakeFillet` takes a _shell_ or a _solid_ as shape and the referenced _edge_ has to be one of the _solid_ or the _shell_ provided, not the one shared by the two _faces_ used to create the _shell_. See example 2.

tnakaicode, 2023-11-23:
I tried that, and got failure.

Tanneguydv, 2023-11-23:
It works for me, on 7.7.0 though

[image]

tnakaicode, 2023-11-23:
You rotated the entire box. That same situation box1.
I want you to rotate only one face.

Tanneguydv, 2023-11-23:
> You rotated the entire box. That same situation box1. I want you to rotate only one face.

Ok, sorry I didn't get your initial goal.
The `MakeShell` method results to a shell composed of the two faces but still with 8 edges, the desired fillet edge is duplicated so it won't build.
The solution is to sew the two faces to make the shell using `BRepBuilderAPI_Sewing`. 
Then we have 7 edges in this shell. We just have to look for the good edge of the shell for the fillet, in your case it's the shell's one shared with the common edge of the two faces used to create the shell.

```python
import sys
import numpy as np
from math import cos, pi

from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepFilletAPI import BRepFilletAPI_MakeFillet
from OCC.Core.TopoDS import TopoDS_Shell
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Display.SimpleGui import init_display
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir, gp_Pnt2d, gp_Ax1, gp_Trsf
from OCC.Core.LocOpe import LocOpe_FindEdges
from OCC.Extend.TopologyUtils import TopologyExplorer
from OCCUtils.Construct import make_box
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing

display, start_display, add_menu, add_function_to_menu = init_display()

box1 = make_box(gp_Pnt(0, 0, 0), 20, 20, 20)
box1_faces = list(TopologyExplorer(box1).faces())

# Make Shell by only two faces
box1_shell = TopoDS_Shell()
bild = BRep_Builder()
bild.MakeShell(box1_shell)
for shp in [box1_faces[0], box1_faces[2]]:
    bild.Add(box1_shell, shp)

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(box1_faces[0], box1_faces[2])
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(box1_shell, 0)
fillet.Add(5, fillet_edge)
fillet.Build()
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")
display.DisplayShape(box1_shell, transparency=0.7)

box2_ = make_box(gp_Pnt(40, 0, 0), 20, 20, 20)

# Rotate 30 degree
box2_trsf = gp_Trsf()
box2_trsf.SetRotation(gp_Ax1(gp_Pnt(40, 0, 0), gp_Dir(0, 0, 1)), np.deg2rad(30))
box2_faces = list(TopologyExplorer(box2_).faces())
box2_faces[0].Move(TopLoc_Location(box2_trsf))

# Make Shell by only two faces that are sewed
sew = BRepBuilderAPI_Sewing()
for face in [box2_faces[0], box2_faces[2]]:
    sew.Add(face)
sew.Perform()
sewed =  sew.SewedShape()

sewed_faces = list(TopologyExplorer(sewed).faces())

# Find Edge that the two faces share
find_edge = LocOpe_FindEdges(sewed_faces[0], sewed_faces[1])
find_edge.InitIterator()
common_edge = find_edge.EdgeTo()

# Find Edge of sewed shape that is shared with common_edge
find_edge = LocOpe_FindEdges(common_edge, sewed)
find_edge.InitIterator()
fillet_edge = find_edge.EdgeTo()

display.DisplayShape(fillet_edge)

# Create Fillet of R5 on shared Edge.
fillet = BRepFilletAPI_MakeFillet(sewed, 0)
fillet.Add(5, fillet_edge)
fillet.Build() 
if fillet.IsDone():
    display.DisplayShape(fillet.Shape())
else:
    print("fillet is not done")

display.DisplayShape(sewed, transparency=0.7)

display.FitAll()
start_display()
```
[image]

tnakaicode, 2023-11-23:
Perfect solution.
Thank you.

closed 2023-11-23 by tnakaicode