tpaviot/pythonocc-core #1285: Requesting pythonocc-core anaconda release to target OpenCascade 7.8.0 to fix issue with BRepBuilderAPI_GTransform

manuel-koch, 2023-11-27:
It would be really cool if `pythonocc-core` could create another anaconda release based on OpenCascade 7.8.0.

I have an open issue with `pythonocc-core==7.7.2` that will likely be fixed by using newer OpenCascade C++ code, see [OpenCascade forum](https://dev.opencascade.org/node/84296#comment-25214).

Cross posting the issue here:

```python
scale_transform = gp_GTrsf()
scale_transform.SetVectorialPart(
  gp_Mat(
    1,    0,       0,
    0,    0.1,    0,
    0,    0,       1    )
)
brep_transform_builder = BRepBuilderAPI_GTransform(
    shape, scale_transform, False  # fails too if I use "True" here
)
```

Raises runtime error:
```
RuntimeError: Standard_ConstructionErrorgp_GTrsf::Trsf() - non-orthogonal GTrsf raised from method BRepBuilderAPI_GTransform of class BRepBuilderAPI_GTransform
```

tpaviot, 2023-11-28:
OpenCascade 7.8.0 has not been tagged yet, afaik there are some changes related to memory management, hopefully it wont have too much impact on the python wrapping workflow

stepi9, 2024-02-14:
OCCT 7.8.0 has been released. Do you have an estimate on when it will be ported to pythonocc?