tpaviot/pythonocc-core #1286: Installation outside conda enviroment

AhmedElsayed-17, 2023-11-28:
Is it possible to install it on another Python interpreter outside the anaconda environment?
because I have a Python version installed with a software I'm using and I want to install pythonocc inside it and unfortunately I have to stick with the software's interpreter because it has some libraries made specifically for it.

Equidamoid, 2023-12-20:
From what I see (and currently trying to do myself), it should be possible to just build it with cmake directly. For you the main challenge will likely be convincing cmake to find your special python (pay attention to the `Found Python3:` line in cmake output. 
This may require hardcoding some things in pythonocc cmakelists file, as cmake's findpython scripts sometimes just refuse to comply.

nickger, 2024-01-23:
Would love to see the support of PyPI to use it in requirements.txt

tpaviot, 2026-10-19:
#1502