tpaviot/pythonocc-core #1290: Shape properties using VolumeProperties and VolumePropertiesGK

markkulait, 2023-12-04:
There are two different functions to calculate the volume properties in the BRepGProp (GProp_GProps, GProp_PrincipalProps) class:
- VolumeProperties
- VolumePropertiesGK
Both calculate the shape volume correctly. However, when calculating the inertia properties of the system, it seems that the former produces the correct results but the latter does not. For a box and a cylinder:
Principal moments of inertia
  Cylinder
           Volume: 196349.54084936203
    Ixx, Iyy, Izz: (194317337.4428578, 194291128.82151097, 61359231.51543226)
  Cylinder GK
           Volume: 196349.5408493644
    Ixx, Iyy, Izz: (6.122366410532748e-40, -2.2351741790771484e-07, -3.5762786865234375e-07)
  Box
           Volume: 124999.99999999999
    Ixx, Iyy, Izz: (52083333.33333338, 52083333.33333331, 52083333.333333276)
  Box GK
           Volume: 124999.99999999999
    Ixx, Iyy, Izz: (1.6326310428087332e-40, -8.940696716308594e-08, -8.940696716308594e-08)

tpaviot, 2023-12-04:
see https://dev.opencascade.org/content/difference-between-brepgpropvolumeproperties-and-brepgpropvolumepropertiesgk

Both methods use different algorithms. GK means Gauss-Kronrod integration. To be rigourous, each algorithm results should be compared to the theoretical solution for volume/Ixx/Iyy/Izz.

tpaviot, 2023-12-04:
```python
>>> print(brepgprop.VolumePropertiesGK.__doc__)

        Parameters
        ----------
        S: TopoDS_Shape
        VProps: GProp_GProps
        Eps: float (optional, default to 0.001)
        OnlyClosed: bool (optional, default to Standard_False)
        IsUseSpan: bool (optional, default to Standard_False)
        CGFlag: bool (optional, default to Standard_False)
        IFlag: bool (optional, default to Standard_False)
        SkipShared: bool (optional, default to Standard_False)

        Return
        -------
        float
```
Did you try to tune the parameters for this function?