tpaviot/pythonocc-core #1297: close open shell boundary

andysfd, 2024-01-09:
Hello

I imported a Step file representing an open shell (<class 'TopoDS_Shell'>).
I now want to find a correct and easy method to close the two existing holes. I searched several documentations but couldn't find a function for this purpose. 

[image]

Furtherone i need to check points inside and outside the shell and the distances, which i successfully accomplished with BRepClass3d_SolidClassifier and BRepExtrema_DistShapeShape. However to get correct results at the beginning of the shell i need to properly close it.

[image]

Thanks in advance for any help!
Wish you all a happy new year

tpaviot, 2024-01-09:
For each red wire, you can build a face using the `BRepBuilderAPI_MakeFace` class:
```python
face_1 = BRepBuilderAPI_MakeFace(red_wire_1).Face()
face_2 = BRepBuilderAPI_MakeFace(red_wire_2).Face()
```

In order to get those wire, use the TopologyExplorer class:
```python
from OCC.Extend.DataExchange import read_step_file
from OCC.Extend.TopologyUtils import TopologyExplorer
step_shp = read_step_file("yourstepfile.stp")
wires = TopologyExplorer(step_shp).wires()
```
and loop over wires

andysfd, 2024-01-09:
so i could extract the vertices of the openings and construct my own faces?

But isn't there a robust solution? i do not know how to detect the edges / vertices associated to the openings. I thought about using ShapeAnalysis_FreeBounds or something but had no success.

When i simply extract the wires and sew them togeter its still open, meaning the red_wire does not exist yet in the model and needs to be constructed from the existing edges.

andysfd, 2024-01-10:
i managed to do it, although i don't think it is the cleanest solution:

I extracted the faces and used the sewing algorithm in order to detect the boundaries with .FreeEdges.
I then extracted the edges and joined them consecutively to a new face. 

Is there an other algorithm which gives me FreeEdges without the need for resewing of the already existing shell?

````
wires = TopologyExplorer(shp).wires()
faces=[]
for wire in wires:
    faces.append(BRepBuilderAPI_MakeFace(wire).Face())

sew = BRepBuilderAPI_Sewing() 
for face in faces:
    sew.Add(face)
sew.Perform()  

print("Free edge", sew.NbFreeEdges())
free_edges =[]
for i in range(1, sew.NbFreeEdges()+1):
    free_edges.append(sew.FreeEdge(i))
````

[image]