tpaviot/pythonocc-core #1298: Can't create Viewer without connecting to X server.

ovo-Tim, 2024-01-25:
I'm trying to create a viewer without connecting to X server and generate an image for this viewer.
My test code:
``` python
import os
import sys
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
# from OCC.Core.Image import Image_PixMap, Image_Format_RGB
# from OCC.Core.Standard import Standard_Byte
from PySide6.QtCore import Qt
from PySide6.QtGui import QPaintEvent, QResizeEvent

from PySide6.QtWidgets import (
    QApplication,
    QLabel,
    QMainWindow,
    QWidget,
    QVBoxLayout
)

from OCC.Display.OCCViewer import Viewer3d

class canva(QLabel):
    def __init__(self, parent: QWidget | None = None) -> None:
        super().__init__(parent)
        self.display = Viewer3d()
        self.setContentsMargins(0,0,0,0)

    def resizeEvent(self, event: QResizeEvent) -> None:
        self.display.SetSize(self.width(), self.height())
        self.display.View.MustBeResized()
        return super().resizeEvent(event)

    def paintEvent(self, arg__1: QPaintEvent) -> None:
        self.display.Repaint()
        self.display.FitAll()

        byte_image = self.display.GetImageData(self.width(), self.height())
        print(byte_image)
        return super().paintEvent(arg__1)

if __name__ == "__main__":
    app = QApplication(sys.argv)
    win = QMainWindow()
    win.setCentralWidget(canva(win))
    win.show()
    sys.exit(app.exec())
```

But it just outputs `None`.

In fact, I'm trying a new way to display OCCT on wayland. Is it possible to generate an image for the viewer and just display the image?

frmdstryr, 2024-01-29:
You can try using `export QT_QPA_PLATFORM=xcb` then running.

ovo-Tim, 2024-01-30:
Hi, in fact I just don't want occt connect to X server. That is because I want to try a new way to display occt viewer on wayand.

tpaviot, 2024-02-22:
You can use the `OffscreenRenderer class`:

``` python
rom OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Display.OCCViewer import OffscreenRenderer
a_box_shape = BRepPrimAPI_MakeBox(10, 20, 30).Shape()
my_renderer = OffscreenRenderer()
my_renderer.DisplayShape(a_box_shape)
```

before, be sure you defined the `PYTHONOCC_OFFSCREEN_RENDERER_DUMP_IMAGE` env variable:
```
$ export PYTHONOCC_OFFSCREEN_RENDERER_DUMP_IMAGE=1
```
This will generate an image in the current folder:

```bash
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Graphic3d_Camera created
Graphic3d_StructureManager created
Display3d class successfully initialized.
#########################################
OpenGl information:
  GLvendor: Intel
  GLdevice: Mesa Intel(R) Xe Graphics (TGL GT2)
  GLversion: 4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
  GLSLversion: 4.60
  Max texture size: 16384
  Max FBO dump size: 4096x16384
  Max combined texture units: 192
  Max MSAA samples: 16
  Viewport: 640x480
  Window buffer: RGB8 ALPHA0 DEPTH24 STENCIL8
  ResolutionRatio: 1
  FBO buffer: GL_SRGB8_ALPHA8 GL_DEPTH24_STENCIL8
OffscreenRenderer content dumped to /home/.../capture-1-1708618626.jpeg
```

tpaviot, 2024-02-22:
wayland is a bit special, the opencascade internal core needs a major refactoring, there already have been a lot of discussions on this issue tracker.

ovo-Tim, 2024-02-23:
All right, I just want to find a way to make OCCT display on Wayland window. But it seems like it's impossible. I have tried to [ask for help](https://dev.opencascade.org/content/it-possible-native-support-wayland) on OCCT forum, but no one pays attention to it. Maybe I should try VTK.

closed 2024-02-23 by ovo-Tim