tpaviot/pythonocc-core #1305: Images are not exported to the correct size

CHAOSdonut, 2024-03-19:
Using the following code i create an offscreen renderer to create a few folders with renders at different angles. I want these images to be 180x180 so i use `offscreen_renderer.SetSize(180, 180)`

my code:
`import os.path
import random
import sys

from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Display.OCCViewer import Viewer3d
from OCC.Core.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Extend.ShapeFactory import rotate_shp_3_axis

input_dir = "./input/"
output_dir = "./output/"

# create the renderer
offscreen_renderer = Viewer3d()

# by default, the offscreenrenderer size is 640*480
offscreen_renderer.Create()
offscreen_renderer.SetModeWireFrame()
offscreen_renderer.EnableAntiAliasing()
offscreen_renderer.SetSize(180, 180)

def read_step_file(filename):
    # read the STEP file and returns a compound
    step_reader = STEPControl_Reader()
    status = step_reader.ReadFile(os.path.join(input_dir + filename))

    if status == IFSelect_RetDone:  # check status
        failsonly = False
        step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
        step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)
        step_reader.TransferRoot(1)
        a_shape = step_reader.Shape(1)
    else:
        print("Error: can't read file.")
        sys.exit(0)
    return a_shape

def random_rotate_shape(shape):
    angle_x = random.random() * 360
    angle_y = random.random() * 360
    angle_z = random.random() * 360
    return rotate_shp_3_axis(shape, angle_x, angle_y, angle_z, "deg")

def render_file(filename, amount):
    # create shape from .stl
    shape = read_step_file(filename)

    output_dir_path = os.path.join(output_dir + filename.split(".")[0])
    if not os.path.exists(output_dir_path):
        os.makedirs(output_dir_path)

    for i in range(amount):
        # give the shape a random rotation
        shape = random_rotate_shape(shape)
        # send the shape to the renderer
        offscreen_renderer.DisplayColoredShape(shape, color='WHITE', update=True)

        output_name = os.path.join(output_dir_path + "/" + filename.split(".")[0] + "(" + str(i) + ").png")
        # export the current shape to an image
        offscreen_renderer.ExportToImage(output_name)
        offscreen_renderer.EraseAll()

if __name__ == "__main__":
    for filename in os.listdir(input_dir):
        render_file(filename, 5)
`

all images generated with this code are 196x219 not 180x180

using `offscreen_renderer.SetSize(1920, 1080)`
results in images that are 1936x1119

tpaviot, 2024-03-21:
Thank you for you report, I never measured the resulting image size and missed that point

tpaviot, 2024-05-23:
Tested on my machine, images are exported with the correct size

[image]