tpaviot/pythonocc-core #1307: ImportError: cannot import name 'TopTools_ListOfShape' from 'OCC.Core.TopTools' after build from source

zhouyc98, 2024-03-25:
I have installed pythonocc from source following the [INSTALL.md](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md). (I'm using macOS so I adjust some config and finally get it installed).
After I installed, I run the [simple test](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md#simple-test) and it is ok; however, the [unit test](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md#test) (test/run_tests.py) is failed and gives the following error message:
```
Traceback (most recent call last):
  File "/Users/ZhouYucheng/Downloads/pythonocc-core/test/run_tests.py", line 22, in <module>
    import core_wrapper_features_unittest
  File "/Users/ZhouYucheng/Downloads/pythonocc-core/test/core_wrapper_features_unittest.py", line 100, in <module>
    from OCC.Core.TopTools import (
ImportError: cannot import name 'TopTools_ListOfShape' from 'OCC.Core.TopTools' (/opt/homebrew/lib/python3.9/site-packages/OCC/Core/TopTools.py)
```
I opened the installed "OCC/Core/TopTools.py" file and found that it has nothing matched `TopTools_ListOfShape`. But it do have a class named `TopTools_ListOfListOfShape`. That's weird. Is it a typo bug?  

I'm using macOS 14.3, pythonocc-7.7.2 (I git cloned this repo and checkout to c8f1d571 which is tagged with 7.7.2), opencascade-7.7.2_3, and SWIG-4.2.1.

zhouyc98, 2024-03-26:
I found it is bug relevant to SWIG version (like #1161). I finally get it worked by installing pythonocc-7.4.1, OCC-7.4.0, and SWIG-3.0.12

tpaviot, 2024-03-26:
If you want to compile the current master branch, swig-4.1.x is required, no need to downgrade to previous pythonocc releases

closed 2024-03-26 by tpaviot

zhouyc98, 2024-03-26:
@tpaviot Thank you for your reply. I now tried swig-4.1.1 with the latest pythonocc and it worked! Thanks!

tpaviot, 2024-05-21:
Fixed in current master branch