tpaviot/pythonocc-core #1308: how to display utf-8 text

ycyang77, 2024-03-27:
How do I display utf-8 text using the DisplayMessage method?

tpaviot, 2024-03-27:
utf8 is supported. Can you please post an example and a screenshot

ycyang77, 2024-03-27:
[image]

Thank you~

ycyang77, 2024-03-27:
[image]

Thank you~

ycyang77, 2024-03-27:
[image]

Thank you~

tpaviot, 2024-04-13:
You did it the right way. Be sure you have the latest version installed.

[image]

closed 2024-04-13 by tpaviot