tpaviot/pythonocc-core #1316: Prototype of GeomAPI_PointsToBSplineSurface.Interpolate

Squirrel8475, 2024-04-27:
Hello all,

I would like to test the interpolation algorithm within the `GeomAPI_PointsToBSplineSurface` class. The [documentation](https://dev.opencascade.org/doc/refman/html/class_geom_a_p_i___points_to_b_spline_surface.html) explains how to switch between the algorithms:
* _use an init function to define and build the bspline surface **by approximation**_. 
This method works fine, the structure is below. But I would like to test interpolation and not approximation, so I have to go to the next point of the documentation.

```
from OCC.Core.TColgp import TColgp_Array2OfPnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSplineSurface
an_array2  = TColgp_Array2OfPnt(1,5,1,5) 
a_maker = GeomAPI_PointsToBSplineSurface(an_array2)
```

* _use an interpolate function to define and build the bspline surface **by interpolation**_
I applied this method to a cloud of points, but the problem I have can be reproduced without, simply using following minimal example:
```python
from OCC.Core.TColgp import TColgp_Array2OfPnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSplineSurface
array2  = TColgp_Array2OfPnt(1,5,1,5) 
a_maker = GeomAPI_PointsToBSplineSurface.Interpolate(array2,False)
````
This code produces the error message:
```python
TypeError: Wrong number or type of arguments for overloaded function 'GeomAPI_PointsToBSplineSurface_Interpolate'.
  Possible C/C++ prototypes are:
    GeomAPI_PointsToBSplineSurface::Interpolate(TColgp_Array2OfPnt const &,Standard_Boolean const)
    GeomAPI_PointsToBSplineSurface::Interpolate(TColgp_Array2OfPnt const &,Approx_ParametrizationType const,Standard_Boolean const)
    GeomAPI_PointsToBSplineSurface::Interpolate(TColStd_Array2OfReal const &,Standard_Real const,Standard_Real const,Standard_Real const,Standard_Real const)

````

And I do not understand why, so any advice appreciated. Thank you in advance.

jf---, 2024-04-27:
> array2  = TColgp_Array2OfPnt(1,5,1,5) 

here you are creating an empty array of `gp_Pnt` instances

use 
``` array2.SetValue(1,1,gp_Pnt())```

to fill out the values in the allocated array

Squirrel8475, 2024-04-27:
This is a minimal example, not my real case. As explained, the real case is based on a cloud of points. The points and the array are well defined because method #1 is working fine and produce a surface as expected.

Squirrel8475, 2024-04-28:
I was confused by the documentation saying the *init* method is just for one algorithm. Both algorithms need the *init* method to instantiate the class as usual in python, so using following structure works:
```python
from OCC.Core.TColgp import TColgp_Array2OfPnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSplineSurface
array2  = TColgp_Array2OfPnt(1,5,1,5) 
a_maker=GeomAPI_PointsToBSplineSurface()
a_maker.Interpolate(array2,False)
```

Please closed the ticket, thank you very much.

closed 2024-05-10 by tpaviot