tpaviot/pythonocc-core #1317: Generating continuous surface from points

kalosu, 2024-04-28:
Hello there pythonocc community,

I am new to this community and maybe what I am asking is really basic. However, I was not able to find any reference example for what I want to do.

I have a set of coordinate points (x,y,z) which are supposed to represent a continuous surface. From these points, I would like to generate a STEP file.

Is it possible to do this using pythonocc? (i.e, via generating a NURBS surface from the discrete points that I have)

Is there maybe some reference on how to do these two things?

1. First generate a NURBS surface from my discrete points.
2. Then save this NURBS surface into a STEP file.

Any comments and help will be appreciated!

tnakaicode, 2024-05-18:
I seem this is not the method you are aiming for, but it is possible to create a triangular face between neighboring points using the Delauney method. However, there are the following problems.

1. Because the planes are tacked together, the surface is not curved.
2. If the point cloud is not convex, the boundary parts are connected.
3. If the point cloud is not based on a certain plane (curved surface like a sphere), as in Bunny's example, the Delauney method needs to be re-examined.
4. If the number of points increases, the calculation becomes heavier

To construct a surface, it is necessary to create an algorithm that 1) constructs a surface from a set of partial points using B-Spline, etc., and 2) connects the surfaces so that the boundary parts do not become discontinuous. I do not know such an algorithm.

```python
import numpy as np

from OCC.Core.gp import gp_Pnt
from OCCUtils.Construct import make_polygon

from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

from scipy.spatial import ConvexHull, Delaunay, voronoi_plot_2d

# https://github.com/tpaviot/pythonocc-demos/blob/master/assets/models/bunny.pcd
pcd_file = "bunny.pcd"
dat = np.loadtxt(pcd_file, skiprows=10)

cov = Delaunay(dat[:, 0:2])
print(cov.simplices)
# print(cov.vertices)

for xyz in dat:
    display.DisplayShape(gp_Pnt(*xyz))

for ixyz in cov.simplices:
    lxyz = dat[ixyz]
    display.DisplayShape(make_polygon(gp_Pnt(*p) for p in lxyz))

display.FitAll()
start_display()
```

[image]

kalosu, 2025-04-17:
Hello again,

I wonder if there is any solution to this at the end? So far I have always used the ```GeomAPI_PointsToBSplineSurface``` interface to generate a Spline surface and then export this as a STEP file.

The thing is, as far as I understand it, it is need to have my input coordinates (x,y,z) in a structured grid. What about the case of an unstructured grid of points, i.e, a point cloud? Is there any way to find the best possible smooth spline fit to the point cloud using pythonocc?