tpaviot/pythonocc-core #1321: Issue with building

sushrut5, 2024-05-10:
Unable to build the latest on master. It seems math_VectorBase.hxx is missing?

tpaviot, 2024-05-11:
Can you please post the full log

tpaviot, 2024-05-13:
Everything is fine on the azure build pipeline, and on my machine, please be sure you correctly follow compilation guidelines at https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md

closed 2024-05-13 by tpaviot

sushrut5, 2024-05-15:
Sorry for the delay and thanks for this work.

The issues I faced seemed like stemming from upgrade to occ 7.8.2 while the README was still asking for 7.7.x

When I went back and checked out the commit 8f3e628, the same development pipeline I was using worked fine. Until that I got errors about various missing files such as the one mentioned in the first comment or another one related to Standards when I tried a commit one older than the head.