tpaviot/pythonocc-core #1326: Suggestions on a project with PythonOCC

Nadjei74, 2024-05-15:
Please I need suggestions on how I can do this with PythonOCC. I have a skinned surface that I want to attach a feature to. Because I want the feature to also have the curvature of the surface, I first want to create the profile of the feature(for example a box) on a plane, then project it to the surface to attain the curvature. After that I extrude the profile from the surface to attain my feature.  I have attached a picture below of what I mean.
Thanks in advance for the help.
[image]

tpaviot, 2024-05-25:
This issue is too general. Please insert any code snippet that can be tested if you need any help.

Nadjei74, 2024-05-25:
So I tried it using boolean section, where I first extrude a the profile of the feature through the skinned surface and with the section I am able to get the profiles on the Skinned surface. The problem now is how to access the profiles on the surface now.

This is how my code looks like (the part that does the sectioning and the part that tries to access these profiles. 

[testnew2.txt](https://github.com/tpaviot/pythonocc-core/files/15441419/testnew2.txt)
 I tried to use the "sectioning.SectionEdges()" but I am unable to iterate through the edges created because the "ToptoolslistofShapeiterator" is not wrapped up in the PythonOCC 7.7.2 version.
[image]
The profiles on the surface are what I am trying to make wire out of.

tpaviot, 2024-05-25:
use `TopTools_ListIteratorOfListOfShape`

Nadjei74, 2024-05-26:
It is not wrapped up in the 7.7.2. Has it been changed in the 7.8.1?

tpaviot, 2024-05-26:
Are you sure its not in 772? I confirm it is available in 781

Nadjei74, 2024-05-26:
[image]
This is a picture I just took

Nadjei74, 2024-05-26:
For updating, I tried 
"conda create --name=pyoccenv python=3.10
source activate pyoccenv
conda install -c conda-forge pythonocc-core=7.8.1" 

from the Readme page but it doesn't seem to work

tpaviot, 2024-05-27:
what do you mean with "doesn' seem to work?". Be careful, you need `TopTools_ListIteratorOfListOfShape`, check the spelling

Nadjei74, 2024-05-28:
No I'm talking about installing pythonOCC 7.8.1 using the conda package.

tnakaicode, 2024-06-04:
BRepAlgoAPI_Section.SectionEdges() returns TopoTools_ListOfShape, and we need another conversion method for TopTools_ListIteratorOfListOfShape.

We need the method to convert TopoTools_ListOfShape to TopTools_ListIteratorOfListOfShape, or TopoTools_ListOfShape must be made Iterable.

```python
display, start_display, add_menu, add_function_to_menu = init_display()

shp = read_step_file("../assets/models/face_recognition_sample_part.stp"
                     # os.path.join("..", "assets", "models", "face_recognition_sample_part.stp")
                     )

def section():
    #orign_profile = create_original(filename)
    sections = []
    for body in range(3):
        section_shp = BRepAlgoAPI_Section(shp, gp_Pln(gp_Pnt(0,0,body+1), gp_Dir(0,0,1)), False)
        section_shp.ComputePCurveOn1(True)
        section_shp.Approximation(True)
        section_shp.Build()
        section_edge = section_shp.SectionEdges()
        print(section_edge.Size())
        for i in section_edge:
            print(i)
        sections.append(section_shp)
```

[image]

tpaviot, 2024-06-04:
@tnakaicode to loop over a `TopoDS_ListOfShape` you need to instanciate a `TopoDS_ListIteratorOfListOfShape`, for example:
```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.TopTools import TopTools_ListOfShape, TopTools_ListIteratorOfListOfShape

shp1 = BRepPrimAPI_MakeBox(10, 20, 30).Shape()
shp2 = BRepPrimAPI_MakeBox(20, 30, 40).Shape()

los = TopTools_ListOfShape()
#print(dir(los))
los.Append(shp1)
los.Append(shp2)

it = TopTools_ListIteratorOfListOfShape(los)

while it.More():
	next_shp = it.Value()
	print(next_shp)
	it.Next()
```
output:
```bash
$ python tst_issue_1326.py 
<class 'TopoDS_Solid'>
<class 'TopoDS_Solid'>
```

Nadjei74, 2024-06-05:
Just installed the pythonOCC core781 from conda forge and the "TopTools_ListIteratorOfListOfShape" is still not wrapped.
[image]

tnakaicode, 2024-06-06:
TopTools_ListIteratorOfListOfShape          --> not defined in TopTools.pyi, but defined in TopTools.py
TopTools_ListIteratorOfListOfListOfShape --> declared not to be defined in TopTools.pyi

we need TopTools_ListIteratorOfListOfShape.

Nadjei74, 2024-06-06:
Sorry but I don't understand what you mean @tnakaicode

tpaviot, 2024-06-06:
@Nadjei74 don't be confused between `TopTools_ListIteratorOfListOfShape` and `TopTools_ListIteratorOfListOfListOfShape`, it is not the same thing.

@tnakaicode comment is pretty clear. The code snippet I posted works on 7.8.1. 7.8.1 can be installed easily using conda-forge. It's up to you right now @Nadjei74 . I thnk the issue can be closed, because the answer got its reply.

closed 2024-06-06 by tpaviot

Nadjei74, 2024-06-07:
It seems you guys are not getting what I am saying. I get that they are different. What I'm saying is, I ran the example code you @tpaviot pasted here but I am still getting an error for the TopTools_ListIteratorOfListOfShape.

Nadjei74, 2024-06-07:
Thanks for the help