tpaviot/pythonocc-core #1336: XCAFDoc_Dimension.DownCast  returned a result with an exception set

aruig666, 2024-05-24:
I think it's possible because TDF_Attribute No constructor defined

`
        doc = TDocStd_Document("MDTV-CAF")
        # self.doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))
        m_dimTolTool = XCAFDoc_DocumentTool.DimTolTool(doc.Main())
        step_reader = STEPCAFControl_Reader()
        step_reader.ReadFile(step_filename)
        step_reader.Transfer(doc)
        root_labels = TDF_LabelSequence()
        m_dimTolTool.GetDimensionLabels(root_labels)
        if root_labels.Length() > 0:
            for i in range(root_labels.Length()):
                label = root_labels.Value(i + 1)
                baseAttr = Standard_Transient()
                aDimAttr = XCAFDoc_Dimension()
                if label.HasAttribute():
                    label.FindAttribute(XCAFDoc_Dimension.GetID(), XCAFDoc_Dimension.DownCast(baseAttr))
`
[image]

tpaviot, 2024-05-25:
Can you please post a code that can be run (add all the required imports, insert the example file as a zip archive)

aruig666, 2024-05-27:
@tpaviot It is the whole code.
`
from OCC.Core.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCC.Core.STEPCAFControl import STEPCAFControl_Reader
from OCC.Core.Standard import Standard_Transient
from OCC.Core.TCollection import TCollection_ExtendedString
from OCC.Core.TDF import TDF_LabelSequence, TDF_Attribute
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool_ColorTool, XCAFDoc_DocumentTool_ShapeTool
from OCC.Extend.TopologyUtils import TopologyExplorer

from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool, XCAFDoc_Dimension
from OCC.Core.Geom import Geom_BSplineCurve

if __name__ == "__main__":
    step_filename = r"C:\Users\agem\Desktop\aaapmi_stp.stp"# import a step with a dimension pmi
    doc = TDocStd_Document("MDTV-CAF")
    m_dimTolTool = XCAFDoc_DocumentTool.DimTolTool(doc.Main())
    step_reader = STEPCAFControl_Reader()
    step_reader.ReadFile(step_filename)
    step_reader.SetColorMode(True)
    step_reader.SetGDTMode(True)
    step_reader.SetSHUOMode(True)
    step_reader.SetMatMode(True)
    step_reader.SetNameMode(True)
    step_reader.SetLayerMode(True)
    step_reader.SetPropsMode(True)
    step_reader.SetViewMode(True)
    step_reader.Transfer(doc)
    root_labels = TDF_LabelSequence()
    m_dimTolTool.GetDimensionLabels(root_labels)
    if root_labels.Length() > 0:
        for i in range(root_labels.Length()):
            label = root_labels.Value(i + 1)
            baseAttr = Standard_Transient()
            aDimAttr = XCAFDoc_Dimension()
            if label.HasAttribute():
                label.FindAttribute(XCAFDoc_Dimension.GetID(), XCAFDoc_Dimension.DownCast(baseAttr)) # the error line. It is also wrong to ues Attribute.DownCast(baseAttr)
            aobj = aDimAttr.GetObject() # in 7.5.1,  could not get the object of attribute
`
Output error message：
`
SystemError: <built-in function Handle_XCAFDoc_Dimension_DownCast> returned a result with an exception set
`
Their inheritance relationship is XCAFDoc_Dimension to TDF_Attribute to Standard_Transient 
 The second input parameter of the label.FindAttribute is of type  TDF_Attribute
in pythonocc-core 7.7.2   
 Attribute.DownCast(baseAttr) or XCAFDoc_Dimension.DownCast(baseAttr) return error 
in pythonocc-core 7.5.1   
Attribute.DownCast(baseAttr) return none and label.FindAttribute(XCAFDoc_Dimension.GetID(), XCAFDoc_Dimension.DownCast(baseAttr)) executed successfully but could not get the object of attribute

tpaviot, 2024-05-27:
insert the aaa_pmi step file, the same you're using

aruig666, 2024-05-28:
[attachFiles.zip](https://githu.com/tpaviot/pythonocc-core/files/15464321/attachFiles.zip)
@tpaviot  I upload my test files.  aaapmi_stp.stp has a dimension pmi as figure:
[image]
nist_ctc_05.stp has many different pmi, and you can find other test step files in:
[https://www.nist.gov/services-resources/software/step-file-analyzer-and-viewer](url)

tpaviot, 2024-05-28:
ok, I think this comes from a missing wrapping method for FindAttribute https://github.com/Open-Cascade-SAS/OCCT/blob/master/src/TDF/TDF_Label.hxx#L147

aruig666, 2024-05-30:
Thanks, how can I update the local  wrapping method?