tpaviot/pythonocc-core #1338: 7.8.1 PackagesNotFoundError

stephensmitchell, 2024-05-26:
Is 7.8.1 published to Conda?

> Channels:
>  - conda-forge
>  - defaults
> Platform: win-64
> Collecting package metadata (repodata.json): done
> Solving environment: failed
> 
> PackagesNotFoundError: The following packages are not available from current channels:
> 
>   - pythonocc-core=7.8.1*
> 
> Current channels:
> 
>   - https://conda.anaconda.org/conda-forge
>   - defaults
> 
> To search for alternate channels that may provide the conda package you're
> looking for, navigate to
> 
>     https://anaconda.org
> 
> and use the search bar at the top of the page.

tpaviot, 2024-05-26:
They're coming, 7.8.1 has bien tagged but i've been waiting for conda packages to be available before making any announce

tpaviot, 2024-05-26:
packages for 7.8.1 are available on conda-forge channel

closed 2024-05-26 by tpaviot