tpaviot/pythonocc-core #1339: Is it planned to build pythonocc-core for linux-arm64 ?

Spoutnik97, 2024-05-26:
I would like to install the library on a docker image build with my Macbook Pro M3.
Currently, the conda installer does not find pythonocc-7.7.2 because it is not available on conda-forge.

Should possible to build the library for linux-arm64 ? Or is there another solution ? 
I could build it myself, but if the need is shared by someone else, we could share ressources and time

tpaviot, 2024-05-26:
there are binaries for osx-arm64 on conda, but not linux-arm64. I don't know if there are plans on the conda-forge side to provide linux-arm64 packages. You can try it by yourself and report here your results. I think that the pythonocc-binderhub docker file is a good place for you to start with https://github.com/tpaviot/pythonocc-binderhub/blob/build-master/Dockerfile

Spoutnik97, 2024-06-04:
Hi @tpaviot !
I tried to build the docker image. It works ! 

But the last command failed while executing python tests

```
ERROR collecting test_mypy_classic_occ_bottle.py _______________
0.380 ImportError while importing test module '/opt/pythonocc-core/test/test_mypy_classic_occ_bottle.py'.
0.380 Hint: make sure your test modules/packages have valid Python names.
0.380 Traceback:
0.380 /opt/conda/lib/python3.11/importlib/__init__.py:126: in import_module
0.380     return _bootstrap._gcd_import(name[level:], package, level)
0.380 test_mypy_classic_occ_bottle.py:22: in <module>
0.380     from [OCC.Core.gp](http://occ.core.gp/) import (
0.380 E   ModuleNotFoundError: No module named 'OCC'
```

Any idea where this could come from?

I will test the OCCT compiled lib to see if it seems ok, and will keep you informed

jenkoj, 2024-06-04:
Hi, the latest version **7.8.1** on conda-forge seems to be corrupt. You cant import the module after instalation.

This Dockerfile works for me on my M2:

```
FROM continuumio/miniconda3:22.11.1 # Feel free to use the latest version

ENV PATH /opt/conda/bin:$PATH

# Install dependencies
RUN apt-get update --fix-missing && \
    apt-get install -y wget bzip2 ca-certificates curl git libc6 && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Update conda
RUN conda update -n base -c defaults conda

WORKDIR /app

# Install pythonocc-core

# Ubuntu
#RUN wget https://anaconda.org/conda-forge/pythonocc-core/7.7.0/download/linux-64/pythonocc-core-7.7.0-py310he34791d_1.conda
#RUN conda install pythonocc-core-7.7.0-py310he34791d_1.conda

# OSX ARM64
RUN wget https://anaconda.org/conda-forge/pythonocc-core/7.7.2/download/osx-arm64/pythonocc-core-7.7.2-py310hbc42c22_1.conda
RUN conda install pythonocc-core*

RUN rm pythonocc-core*

.
.
.
```

tpaviot, 2024-06-05:
@jenkoj what do you exactly mean with "7.8.1 seems to be corrupt"? Can you please post the full trace when installing 7.8.1 from conda-forge channel?

tpaviot, 2024-06-05:
> ModuleNotFoundError: No module named 'OCC'

@jenkoj that means pythonocc is not installed in a known python path

jenkoj, 2024-06-05:
@tpaviot  and @Spoutnik97, my mistake. I thought docker was using osx-arm64 kernel. So please ignore my Dockerfile above. While it does install the package, it doesnt work. To avioid confusion, I am deleting that comment. 

@tpaviot when installing directly using wget, I noticed that **7.2.2** has compatible python version in its name, whereas **7.8.1** does not. So **7.8.1** is not corrupted, its just hard to find the right version for you python version.

**As you suggested, only solution for running occ in container on mac is to build the occ from source.**

Spoutnik97, 2024-06-05:
> > ModuleNotFoundError: No module named 'OCC'
> 
> 
> 
> @jenkoj that means pythonocc is not installed in a known python path

But it should work if it work for you. Where could be the problem ? Bad path ? Bad compilation ?